z88dk/z88dk #155: sccz80: Support --constseg, --codeseg support

suborb, 2017-04-10:
To define the names of the sections used by the compiler.

Might as well add: --bssseg and --dataseg to complete the set.

aralbrec, 2017-04-18:
If sccz80 is given section names on the compile line like this:

zcc +zx -v -c -clib=new eliza.c --codeseg=THIS --constseg=THIS --dataseg=THIS --bssseg=THIS

everything is ok.  Code and data is placed in section "THIS".

If sccz80 is given section names on the compile line like this:

zcc +zx -v -c -clib=new eliza.c --codesegTHIS --constsegTHIS --datasegTHIS --bsssegTHIS

the section names used are incorrect :- they become "--codesegTHIS", "--constsegTHIS", "--datasegTHIS" and "--bsssegTHIS".  Ie the command line option is not removed from the section name.

suborb, 2017-04-19:
Right, okay, sorry I'll add in some logic to cope with that 3rd way of defining that option. At the moment sccz80 accepts:

--bssseg=THIS
--bssseg THIS

So what's happening in your case is that the name of the code segment becomes "--constsegTHIS" and the BSS segment is not being changed.

aralbrec, 2017-04-19:
I must have misread one of the updates -  thought you had added this form for compatibility with sdcc.  It's a good idea anyway since I am not sure how easy it is to change sdcc's behaviour.  The segment options are in a global options structure that affects all backends.

suborb, 2017-04-19:
I think I must've thought the sdcc form was a typo since it's a bit weird.

Anyway, it's now supported so there's 6(!) ways of defining the option now:

-bssseg=VALUE
--bssseg=VALUE
-bssseg VALUE
--bssseg VALUE
-bsssegVALUE
--bsssegVALUE

closed 2017-04-19 by suborb

aralbrec, 2017-04-19:
It will actually support two forms "--codeseg FOO" and "--codesegFOO" but we have to give "--codesegFOO" to zcc so that zcc forwards both the option and the FOO to sdcc.  With a space in there zcc would treat the FOO as a separate file.   We could give zcc '-Cc"--codeseg FOO"' but that is far uglier IMO.

commit baecbd35 (2017-04-10): Merge pull request #158 from z88dk/feature/section_names

Fix #155: sccz80: Allow section names to be configured on the command line.
commit 4e0adbd9 (2017-04-19): Fix #155: now accept the value following on directly from the option