z88dk/z88dk #156: zsdcc: add --bssseg and --dataseg to control destination bank for bss and data sections

aralbrec, 2017-04-10:
To keep pace with sccz80.
Complete file level bank assignment is one ingredient in the generation of bankswitched code.

aralbrec, 2017-04-12:
sccz80 is allowing "--constseg=BANK_02" but zsdcc must do "--constsegBANK_02" which sccz80 will not accept.

The equals is much better so look into whether zsdcc can be modified to do things that way too.  It should just be down to eliminating punctuation characters after an option has been recognized.

suborb, 2017-04-13:
I do want to sort out the sccz80 argument parsing so that it accepts different variants of commands, so that long arguments can either use '=' or have the value as then next argv.

It's complicated by having long arguments with both a single '-' and '--' of course.

I'll see what I can do this weekend though.

suborb, 2017-04-22:
I'm going to add another one to sccz80 to control the initialisation section which currently has the
name: code_crt_init

So, --initseg=XXXX

aralbrec, 2025-06-26:
This was causing way too much friction when using zsdcc to target banked environments.  It occurred to me, this doesn't have to be fixed in zsdcc and can instead be fixed by text substitution on the zsdcc output from within zcc.

I'm going to commit this fix shortly and hopefully you guys can be alert for any issues.  I don't think there will be any.

zcc will intercept the following options:

```
--allseg=NAME_ALLSEG
--codeseg=NAME_CODESEG
--constseg=NAME_CONSTSEG
--dataseg=NAME_DATASEG
--bssseg=NAME_BSSSEG
```

`--allseg` is new and makes all the sections the same so it's a single param that replaces potentially four individual ones for the common case of throwing the entire compiler output into the same section.

If sccz80 is doing the compiling, zcc just passes these parameters to the sccz80 compile line.

If zsdcc is doing the compiling, zcc processes the output after the c_sdccopt1 rule file has been applied to replace compiler sections "code_compiler", "rodata_compiler", "data_compiler" and "bss_compiler" with the new section names.  It's important that c_sdccopt1 be applied first because these are copt rules for translating the sdcc areas to z88dk's compiler section names.

I didn't touch sccz80 `--initseg` as that has special purpose so it's not part of `--allseg`.

I also didn't touch the ez80 compile path as that seems a work in progress and doesn't do section assignment yet.

aralbrec, 2025-06-26:
I did notice that sccz80 seemed to place const data that is not initialized into the wrong segment:

```
const unsigned int a;
```

I'd expect that to have 0 value and placed in the consteg (rodata_compiler).  Instead sccz80 places that in bss.

commit d203eb07 (2025-06-26): Fix #156 : Allow ZDSCC to correctly generate code and data into all sections
closed 2025-06-26 by aralbrec