z88dk/z88dk #162: Philips Videopac C7420 not fully tested

zx70, 2017-04-12:
Experimentally discovered:

        ;ld      hl,-64
        ;add     hl,sp
        ;ld      sp,hl
..needs to be commented out on the startup code, and on fputc_cons:

; ODDLY THIS DOESN'T WORK !   (O2EM emulator problem ?)
;        ld      hl,2
;        add     hl,sp
;        ld      a,(hl)

; THIS WORKS !!
	pop bc
	pop hl
	ld a,l
	push hl
	push bc

I strongly suspect the ADD HL,SP instruction not being correctly emulated on O2EM, a better test environment needs to be defined, and the target port needs to be confirmed as correctly working.

suborb, 2017-04-13:
Is there a link to the source for the emulator version with the 7420 support? All I can find are windows binaries.

Without add hl,sp then I'm pretty confident that no C code will work! Even the sdcc builds will call library routines that use add hl,sp

zx70, 2017-04-14:
I found the C7420 listed on the current MAME versions but I haven't tried it (need to deal with roms and configuration).
Appmake produces valid wav files, so if the MAME emulation is complete they could be attached with the "-cass1"  command line option.   I consider this low priority, though..  I just opened the issue not to forget it !

zx70, 2017-05-03:
I may be wrong, but MAME doesn't seem to be ready to support C7420 and cassette tape transfers.
I found a recent copy of the particular O2EM variant supporting the z80cpu, ([http://guttenbrunner.com/videopac/o2em120src.zip]) it uses the Marat Fayzullin's Z80 emulation engine.   I hardly believe it can be broken ..