z88dk/z88dk #163: Amstrad CPC targets: stability problems ?

zx70, 2017-04-12:
We seem to have stability problems with the CPC target(s).
I could see output working by a sequence of fputc_cons calls, or minimal printf but nothing more, investigation needed.

zx70, 2017-04-12:
Extra comment: tested in on the CPC464 emulation (MESS).
A very short initial part of the programs can be seen on output if I comment out the "EI" instructions on the CRT0 file.

aralbrec, 2017-04-12:
I'll have a look at this.  The cpc was fixed recently and should be rock-solid.  I had no trouble running any cpc programs.

zx70, 2017-04-15:
Correct myself.. tested on cpc664 and 6128, not cpc464.  ..is it relevant?

suborb, 2018-02-23:
The 464 was relevant, fixed in e7bbebc0d6f5b6fbfc2ac6dbe8cadfec8878347a

closed 2018-02-23 by suborb

zx70, 2018-02-24:
forgot to close it myself,  well done,  Dominic  👍