z88dk/z88dk #164: ZX81 and Lambda not able to run sdcc compiled code

zx70, 2017-04-12:
Some programs are very close (adv_a fails in an interesting way..)  but the way they use the index registers is creating problems either to sdcc or the glue components.

aralbrec, 2017-04-12:
Try sdcc compiles with "--reserve-regs-iy" to stop sdcc from using IY.  In fact, you will most often get better code out of this when optimization is high with -SO3.

This will not always work -- sdcc will use IY if the stack frame is too large so that IX can't be used to address all local variables.

zx70, 2017-04-14:
"--reserve-regs-iy"  is the only hope to have runnable code on the ZX81 family, the IX register is part of the display driver and it is used by the NMI interrupt code in ROM as a JP vector.
The "IXIY" swap directive on the assembler inverts the index registers to reserve IX in place of IY.   The ADV_A example works well on normal targets also with the IY reservation, so the problem cannot be on the stack frame size.