z88dk/z88dk #167: new c lib: enhance font libraries

aralbrec, 2017-04-15:
While working on the SMS target I realized two things:

* the SMS has no readily available fonts
* although the SMS defines 8x8 characters and can therefore nominally use "standard" 8x8 fonts, it combines colour information with each character such that each character occupies 32 bytes instead of 8.

In order to use these standard 1-bit fonts, I'm going to supply a function that copies these fonts into vram while adding colour information.  This way the SMS can make use of the many 8x8 fonts created on other systems.

There are some things that need to be improved in the new c lib's font libraries to make things easier:

* [ ] Each font needs a start and end address associated with it so that the program can determine how much space the font occupies in memory.

For example:

```
SECTION rodata_font
SECTION rodata_font_8x8

PUBLIC _font_8x8_clairsys
PUBLIC _font_8x8_clairsys

_font_8x8_clairsys:

   BINARY "font_8x8_clairsys.bin"

_font_8x8_clairsys_end:
```

This goes for font4x8, font8x8, and fzx fonts.

* [ ] Headers have to be created so that the C program can reference these fonts.  Right now the terminal drivers are the only users of fonts.
* [ ] We need more fonts.  Only a handful are in the library.
* [ ] Before the font4x8 type goes too far, look into halving the memory occupation of 4x8 by only storing 4 bits wide per character.  Right now, the four bits are repeated twice in each character in order to speed up printing but I think it's more important to save memory space.

suborb, 2017-04-21:
Trawling though images of my old discs at the weekend, I found a disc that had about 50 fonts on there.

Not too certain where they came from or what the quality is like but I'll knock up a quick tool to create pngs of fonts and we can then see.

I seem to remember quite a few fonts being published in Outlet so it might be worth looking there as well.

zx70, 2017-05-08:
I tried to use the  sprite editor to import the pbm pictures and discovered that not all the allegro implementations permit the listed graphics modes.    I quickly hacked the tool to add a native PBM support in case Allegro fails the picture load.
Crystal looks blank and some of the fonts I posted are already in your list.
Please be aware that the "import bitmap" function was already able to parse a SNA file and import the custom font currently configured on the 23606 pointer + the UDG set on bottom of a big square "sprite".

zx70, 2017-05-10:
Sprite editor improved a little more, now it features also (crap) support for multiple elements in BDF and PBM formats.  
I tried it on these fonts:
https://github.com/Tecate/bitmap-fonts/tree/master/bitmap/artwiz/bdf
.. extra editing and a adjustment is needed and the import is not perfect, but I think it is already cool  :)

zx70, 2017-06-05:
Font import and preview functions are now at a very good stage.   Picking patterns here and there I could easily build a personal library of > than 400 fonts (proportional and fixed size), but obviously the web is quite an unlimited resource: font data can be grabbed from Linux, PrintMaster, GEOS, MAME ROMs, screenshots, etc..

A quick and dirty way to use a proportional font on almost all the z88dk targets (using the "putsprite" based implementation for the "x11" library) is explained here:
https://www.z88dk.org/forum/viewtopic.php?pid=14659#p14659

..can it be merged as a common feature in the classic z88dk side ?

feilipu, 2026-10-27:
The fixed-size font libraries now follow the layout asked for here. The 8×8 set is in [`libsrc/font/font_8x8`](https://github.com/z88dk/z88dk/tree/master/libsrc/font/font_8x8) (moved out of `libsrc/_DEVELOPMENT` by [7db80edc60](https://github.com/z88dk/z88dk/commit/7db80edc60c6c8ec0aa53cc20ef375e3c7d7a966)). Each font is a raw image and a ZX7 image, and each module exports a start label and an end label. The same wrappers are used for `libsrc/font/font_4x8` and `libsrc/font/font_8x10`. C names are in [`include/font/font.h`](https://github.com/z88dk/z88dk/blob/master/include/font/font.h). FZX names and the drawing API are in [`include/font/fzx.h`](https://github.com/z88dk/z88dk/blob/master/include/font/fzx.h).

**Start, end, and compressed copies**

- [ada8b35a55](https://github.com/z88dk/z88dk/commit/ada8b35a5530c7c0002024ec3d1c5024733ce1d0) — `#167 spruce up the font directory for the sms target`. Adds the `SECTION` / `BINARY` / `_end` pattern and the first ZX7 copies (ClaireSYS, ClaireSYS bold, ZX, and the default 4×8).
- [29752aee73](https://github.com/z88dk/z88dk/commit/29752aee7305b6bc85b4a33b88391f74c9659791) — exports the 8×8 `_end` labels. The first commit had defined them and published the start label twice.

Later fonts use that same pair of modules. Raw 8×8 images are 768 bytes (ASCII 32 onward, eight bytes per glyph). The twelve ZX7 images are 5413 bytes together.

**SMS 1-bit font copy**

- [61bcfe1b99](https://github.com/z88dk/z88dk/commit/61bcfe1b99626650df63f9b6860363d9b375b8c4) — `sms_copy_font_8x8_to_vram()`, which copies a standard 1-bit 8×8 font into SMS VRAM and adds the colour bytes.

**Headers**

- [c025cdf038](https://github.com/z88dk/z88dk/commit/c025cdf038ab925b393f23c032fd4882e10d97b3) — adds `include/font/font.h` for the 8×8 and 4×8 sets.
- [18793fdb78](https://github.com/z88dk/z88dk/commit/18793fdb78c825be4a08b31951ad64b22e4f3100) — adds Einstein, MSX, PMD85, and SAM.
- [9f68e74cb0](https://github.com/z88dk/z88dk/commit/9f68e74cb0a68ba5a9bf10fe33f74130286c4927) — adds Coleco Adam, the 8×10 set, and the 6×8 names. Those 6×8 names are the same 768-byte 8×8 images, for TMS9918 text mode.
- [4df1fb237a](https://github.com/z88dk/z88dk/commit/4df1fb237ac12dfebcc3f72b94a77ea31f312ba9) — FZX fonts are structures in the undecorated C headers, so a program can name `ff_*` directly.

**Font set**

- [2e657147d6](https://github.com/z88dk/z88dk/commit/2e657147d63c1b1ae811dd0b855e53d1524471e2) — BBC Micro 8×8, raw and ZX7.
- [a4c86851e2](https://github.com/z88dk/z88dk/commit/a4c86851e222cf38d70d7fe1007dbe023d7322ca) and [292f4a5fc0](https://github.com/z88dk/z88dk/commit/292f4a5fc0a78c3fe95a56f727da76a7d87e5b2a) — 4×8 fonts from the 80-column set and from `64.tap`.
- [2553b93b9e](https://github.com/z88dk/z88dk/commit/2553b93b9e158b81c27c570fb0a64f5bce534b3e) — brings those 4×8 fonts into newlib with ZX7 twins, and adds C64, CPC, and VGA 8×8.
- [bf7a605132](https://github.com/z88dk/z88dk/commit/bf7a60513248b03c987ad517d9ae6aca55a3653e) — MSX, Einstein, and PMD85 8×8.
- [cee0aaa2f6](https://github.com/z88dk/z88dk/commit/cee0aaa2f61d8f5f9631f9ad8b804697b2b53709) — SAM 8×8.
- [f927c84dbe](https://github.com/z88dk/z88dk/commit/f927c84dbeec85564302d630d784d9b8ad53ae3f) — Coleco Adam 8×8.
- [0ca9c3caf8](https://github.com/z88dk/z88dk/commit/0ca9c3caf86365a9463e7573676f92e285700c45) — 8×10 fonts for the VG5000, raw and ZX7.
- [71a1e7a0e6](https://github.com/z88dk/z88dk/commit/71a1e7a0e648b5a8c010a25f03dab84147f34704) — FZX proportional fonts. [3657d482b7](https://github.com/z88dk/z88dk/commit/3657d482b7e90e973c42b5aa8482d88f42773c93) adds the Kreative Korp and Seven.FFF sets. `fzx.h` now names 189 fonts.

**Left as they are**

The 4×8 images stay 768 bytes with the glyph nibble stored in both halves of each byte. The 64-column printers (`__zx_printc64`, the ZX 64-column terminal, and the Next 128-column terminal) mask with `$f0` or `$0f` and use that layout. ZX7 is the size reduction those fonts got.

FZX modules have no `_end` label and no ZX7 copy. The uncompressed length is the final offset word in the FZX header: it is relative to that word and points at the byte after the last glyph. That matches every `.fzx` file in the tree.

closed 2026-10-27 by feilipu