z88dk/z88dk #174: z80asm: complete Z180 and Rabbit support

pauloscustodio, 2017-04-16:
Z180 and Rabbit RCM2000 and RCM3000 are partially supported in z80asm.

Missing areas are a complete test of the Z180 opcodes and the Rabbit instructions not existing in Z180.

Complete this support.

This is issue is spun of #135

pauloscustodio, 2017-04-17:
Refer to commit d858404af35f1b852eb1a4b9b085c48f875d628f

pauloscustodio, 2017-04-17:
Extend options for CPU selection 
e29bee72864b1fed6cb79a186567205998776ec5

pauloscustodio, 2017-04-17:
Fix parser for Z180 
1631c64f51c3c0fc974227dae0fa597a9718546b

suborb, 2017-04-28:
I didn't realise you were going all out on Rabbit support!

Most of the non-digi assemblers just accept an altd modifier and then check whether it can be applied to the opcode.

Certainly the altd/ioi prefix is what sdcc will output as necessary.

pauloscustodio, 2017-04-28:
Complete the Rabbit support was on the to-do list anyway. I've added all the new opposes and am now through a second pass checking where ALTD is allowed and adding the corresponding syntax, e.g. ld a',a. I've added some new opcodes, like cpl a; cpl a'; scf f; scf f'
(These are arguable and may not be a good idea).
I did not find a simple understandable syntax to represent opcodes that only change f ', e.g. altd bit 0,b as bit 0,b,f '

I'm going to submit the opcodes lists for peer review before merge.

pauloscustodio, 2017-09-03:
Closed and merged with commit 1833761eca984a9bf05e4b70b478b890c5bf2bdd

closed 2017-09-03 by pauloscustodio

suborb, 2017-09-03:
Thank you very much for this monster effort.

pauloscustodio, 2017-09-03:
You're welcome! It was fun anyway!

suborb, 2017-09-03:
Given the object file are incompatible do we need to mark object files with the CPU and validate at link time? I can foresee issues where someone tries to use a ZXN library on a vanilla Spectrum with odd effects.