z88dk/z88dk #176: sccz80: fails to accept common sizeof idiom

aralbrec, 2017-04-18:
Test program:

```
#include <stdlib.h>

unsigned int *p;

void main(void)
{
	p = malloc(10*sizeof(*p));
}
```

sccz80 produces errors:
zcc +zx -vn -a -clib=new test.c

sdcc accepts it:
zcc +zx -vn -a -clib=sdcc_iy test.c

suborb, 2017-04-19:
I've still got to sort out sizeof() on members of nested structs, but what's there should fix the issue you've experienced.

aralbrec, 2017-04-20:
This may fall into a more complicated case but this one will not work after the fix:

```
unsigned char *quotes[] = {
	"Hello",
	"World",
};

unsigned int num_quotes(void)
{
	return sizeof(quotes)/sizeof(*quotes);
}
```

sccz80 fails:
zcc +zx -vn -a -clib=new test.c

sdcc succeeds:
zcc +zx -vn -a -clib=sdcc_iy test.c

suborb, 2017-04-20:
Good point, I knew there had to be some missing tests! I'll sort that out this evening along with the other ones I've thought about today.

aralbrec, 2017-04-20:
One more case:

```
typedef struct
{
	unsigned char xOffset, yOffset;
	unsigned char tileOffset;
	signed char reservedSprite; //, or -1
}subSprite;

typedef struct
{
	unsigned int x, y;
	unsigned int tile;
	unsigned char nbSprites;
	subSprite	(*sprites)[];
} metaSprite;

void main(void)
{
}
```

sccz80 fails on "subSprite (*sprites)[];":
zcc +zx -vn -a -clib=new test.c

sdcc succeeds:
zcc +zx -vn -a -clib=sdcc_iy test.c

"subSprite (*sprites)[];" should be a pointer to an array of subSprite, ie "subSprite **sprites;".

suborb, 2017-04-20:
That last one is a different issue and code area.  I'll pull it into a different issue. What happens with a size of 0 btw?

Currently sorting out arrays of far pointers so I'll tackle this one later.

suborb, 2017-04-20:
There's one case that's still outstanding, which I'm going to leave until there's some real code that has the problem:

    struct st vals[10];

    sizeof(vals[0].c);   // Parser error

aralbrec, 2017-04-20:
> That last one is a different issue and code area.  I'll pull it into a different issue. What happens with a size of 0 btw?

I'm not sure what you mean by that.  If they put a size in "subSprite (*sprites)[];" I guess the size becomes part of the type and you can only assign arrays with the specific size to subSprite.

suborb, 2017-04-21:
>>That last one is a different issue and code area. I'll pull it into a different issue. What happens with a size of 0 btw?

>I'm not sure what you mean by that. If they put a size in "subSprite (*sprites)[];" I guess the size becomes part of the type and you can only assign arrays with the specific size to subSprite.

I think it was a misinterpretation by me, thinking that a variable number of subSprites were contiguous within the metaSprite structure.

For the sake of reference can you drop in a line indicating how the sprites member is used? The obvious `&struct->sprites[n]` was declined by both clang and sdcc

aralbrec, 2017-04-21:
Here are some examples of how it's used:

```
//first sprite is not defined from metaSprite.x/.y/.tile
//this way I could use a metaSprite at (0,0)
//but its first subSprite at (3,2) for ex
//==> not squared sprites !!
typedef struct
{
	unsigned char xOffset, yOffset;
	unsigned char tileOffset;
	signed char reservedSprite; //, or -1
}subSprite;

//metasprite is a logic sprite
//real SMS sprites are created from meta's subsprites
typedef struct
{
	unsigned int x, y;
	unsigned int tile;
	unsigned char nbSprites;
	//using a pointer to an array allows to store this array on ROM
	//generated by a (coming soon?) tool
	subSprite	(*sprites)[];
} metaSprite;
```

```
void addMetaSprite(metaSprite *msprite) 
{
	unsigned char i;
	unsigned char count = msprite->nbSprites;
	subSprite *sub;

	sub = & (*msprite->sprites)[0];
	for (i = 0; i< count; i++)
	{
		SMS_addSprite(	sub->xOffset + msprite->x,
						sub->yOffset + msprite->y,
						sub->tileOffset + msprite->tile );
		sub++;
	}
}
```

```
static subSprite	subShip[4*2];
static metaSprite 	shipSprite;

	u8 idx = 0;

	subShip[idx].xOffset = 0;
	subShip[idx].yOffset = 0;
	subShip[idx].tileOffset = idx;

	idx++;
	subShip[idx].xOffset = 8;
	subShip[idx].yOffset = 0;
	subShip[idx].tileOffset = idx;

	......

	idx++;
	shipSprite.tile = TILE_SHIP;
	shipSprite.nbSprites = idx;
	shipSprite.sprites =&subShip;
```

closed 2017-04-22 by suborb