z88dk/z88dk #180: sccz80: Defining pointer to array (with no array size)

suborb, 2017-04-20:
Alvin reported in #176 

```
typedef struct
{
	unsigned char xOffset, yOffset;
	unsigned char tileOffset;
	signed char reservedSprite; //, or -1
}subSprite;

typedef struct
{
	unsigned int x, y;
	unsigned int tile;
	unsigned char nbSprites;
	subSprite	(*sprites)[];
} metaSprite;

void main(void)
{
}
```

sccz80 fails on "subSprite (*sprites)[];":
zcc +zx -vn -a -clib=new test.c

sdcc succeeds:
zcc +zx -vn -a -clib=sdcc_iy test.c

"subSprite (*sprites)[];" should be a pointer to an array of subSprite, ie "subSprite **sprites;".

suborb, 2017-04-20:
So, sccz80 thinks that this the definition of sprites is introducing a function pointer.

aralbrec, 2017-04-20:
> I think for this usecase, just a `subSprite *sprites` achieves what's intended - a pointer to an array of subSprite? Whereas `subSprite **sprites` ends up being a pointer to an array of pointers to subSprite.

Yes I think you are right.  It's easy to become messed up when people use unusual declarations.  An array is a fixed memory address so a pointer to one should be a pointer to an array element.

suborb, 2017-04-22:
From the other thread:

Here are some examples of how it's used:

```
//first sprite is not defined from metaSprite.x/.y/.tile
//this way I could use a metaSprite at (0,0)
//but its first subSprite at (3,2) for ex
//==> not squared sprites !!
typedef struct
{
	unsigned char xOffset, yOffset;
	unsigned char tileOffset;
	signed char reservedSprite; //, or -1
}subSprite;

//metasprite is a logic sprite
//real SMS sprites are created from meta's subsprites
typedef struct
{
	unsigned int x, y;
	unsigned int tile;
	unsigned char nbSprites;
	//using a pointer to an array allows to store this array on ROM
	//generated by a (coming soon?) tool
	subSprite	(*sprites)[];
} metaSprite;
```

```
void addMetaSprite(metaSprite *msprite) 
{
	unsigned char i;
	unsigned char count = msprite->nbSprites;
	subSprite *sub;

	sub = & (*msprite->sprites)[0];
	for (i = 0; i< count; i++)
	{
		SMS_addSprite(	sub->xOffset + msprite->x,
						sub->yOffset + msprite->y,
						sub->tileOffset + msprite->tile );
		sub++;
	}
}
```

```
static subSprite	subShip[4*2];
static metaSprite 	shipSprite;

	u8 idx = 0;

	subShip[idx].xOffset = 0;
	subShip[idx].yOffset = 0;
	subShip[idx].tileOffset = idx;

	idx++;
	subShip[idx].xOffset = 8;
	subShip[idx].yOffset = 0;
	subShip[idx].tileOffset = idx;

	......

	idx++;
	shipSprite.tile = TILE_SHIP;
	shipSprite.nbSprites = idx;
	shipSprite.sprites =&subShip;
```

suborb, 2019-01-16:
The decl rewrite back in 2017 banished the compiler error so I think this one can be closed

closed 2019-01-16 by suborb