z88dk/z88dk #184: new c lib: AY-3-8910 drivers

aralbrec, 2017-04-21:
There are several cross development tracker programs that build music for this chip on several targets:

* [Arkos Tracker](http://www.grimware.org/doku.php/documentations/software/arkos.tracker/start)
* [WYZ Tracker](https://github.com/AugustoRuiz/WYZTracker)
* [Vortex Tracker II](http://bulba.untergrund.net/vortex_e.htm)

and for sound effects:

* [AYFX Editor](https://shiru.untergrund.net/software.shtml)

Each of them supply their own z80 code to play their songs.  However, the code is targeted at a narrow range of computers due to hard coding of ports and the code tends to be self-modifying.  This means the songs they produce are not easily ported to other machines and cannot target more than one AY chip at a time, of which there are a few examples.

The goal is to clean up the playing code such that instead of outputting directly to hardware, they fill a data structure at specified address containing AY register values.  A separate routine provided by the platform can then write the AY data to the chip, apply effects or mix output from several player / sound effect sources.

feilipu, 2019-07-05:
Also useful for the RC2014 target using [Ed's YM/AY Sound Card](https://www.tindie.com/products/Semachthemonkey/ym2149-sound-card-for-rc2014-retro-computer/), which is now available from official channels.

suborb, 2020-02-05:
This came up on the forum here: https://www.z88dk.org/forum/viewtopic.php?id=11186

Looking at WYZPlayer it looks there's one routine which needs to be extracted on a per machine basis - it's basically filling a buffer with the registers for the AY. I think Arkos does something similar - they don't look self modifying.

So, I think I'm going to focus on WYZ and create an output routine for supported machines. This then needs to be wrapped up into something easier to use from C - initialisation + hook into interrupts.

feilipu, 2020-02-05:
There's code at [RomWBW for a console bell](https://github.com/wwarthen/RomWBW/blob/master/Source/HBIOS/ay.asm), which might be useable at least to get a beep into the stdio library.

feilipu, 2020-02-06:
And, a version of the [tune ](https://github.com/wwarthen/RomWBW/tree/master/Source/Apps/Tune) program working with hbios api.

suborb, 2021-05-01:
WYZ and Vortex are available in classic across multiple machines. WYZ is available in newlib. 

So I'd say this one is done for the moment.

closed 2021-05-01 by suborb