z88dk/z88dk #192: classic: look into reported problems with sms target

aralbrec, 2017-04-24:
I've had some feedback from users of the classic sms target that there can be graphical glitching when using its API.  This could be down to not disabling interrupts around writes to the vdp command register or the interrupt altering the vdp state such that it interferes with the main program.

And some example programs not working.

suborb, 2017-04-24:
The tiles for the pong examples were corrupted due a misplaced ix save (for sdcc compatibility). 

Putting it in the right place fixes the graphics/restores functionality to what was there at the start of 2016.

suborb, 2017-04-25:
Doing a trawl of the web indicates that:

* We need to do a di/ei around the VDP port rights
* The interrupt routine may be wrong

suborb, 2017-04-25:
The changes in that commit get the .sms files working on the emulators I've tried (previously they weren't) so I think that can be considered successful.

aralbrec, 2017-04-27:
There are a lot of timing violations in the library.
Two successive VDP writes to the data port must be separated by 29 cycles when ints are on.

suborb, 2017-04-28:
I'm confused by this one. The docs say 29T, but online comments say it just needs a nop between the writes.

There's also code in newlib that isn't separated by 29T.

What's the real answer for the separation required?

aralbrec, 2017-04-28:
Timing for the command register does not matter.  Only access via the data register needs to be properly timed.  This makes sense since these are accessing the VDP RAM.

I read somewhere that the users are saying 26T is sufficient instead of 29T but I'm not sure if that is reliable or not.

In devkitSMS there is a concept of "unsafe" and "safe" access.  "Safe" access means 29T separation and therefore code that can run anytime.  "Unsafe" is as fast as you can go but this will only work if the VDP is off (not displaying either via VBI or switch).

sverx, 2017-05-07:
In devkitSMS "safe" access (default) means 26 CPU cycles delay minimum. That value comes from extensive testing made long ago and we still didn't come across a single case where there was corruption or any other problem with that value. :)

aralbrec, 2017-05-07:
Thanks sverx.  Reading through a lot of old postings, it's sometimes hard to identify when something is authoritative or speculative.

aralbrec, 2017-10-22:
I've just been reminded that there are still problems in the classic lib's sms support.

[3d city](https://github.com/z88dk/z88dk/tree/master/examples/sms/3dcity) in the examples directory still cannot compile to functioning code.  The game currently freezes at the splash page.

suborb, 2017-10-28:
This poke made me use 3dcity in my sccz80 branch as a test program. 

I found a couple of code generator issues, but for me, using OpenEmu and the CrabEMU core the game works as expected.

Using TwoMBIT the game just hangs at the splash page. 

Using SMSPlus, the splash page vanishes, the title screen comes up, and then we end up with a corrupted screen.

Unfortunately non of the emulators has a debugger so I've no idea whats wrong, which is a bit awkward given that I've got 3 different conditions.

The last time round I fixed up all the VDP timings and patched a few things around interrupts, however given the variety of conditions I'm wondering if we've got an issue with either VDP setup or interrupt handling.

aralbrec, 2017-10-29:
The emulator to use is Emulicious.  This one is regarded as highly accurate and it has some very good debugger functionality built in.  There's also an "enforce vdp timing constraints" mode and a mode that does not enforce timing constraints.  For me, either one results in the same thing which is hanging after the splash page.  I would have thought successfully running on the other emulators would also happen for Emulicious with timing constraints off.

Anyway with timing constraints on, the emulator will also tell you when violations occur.  So some time spent with this might identify some problems.

suborb, 2017-10-29:
I didn't know about that one, and it's java and runs on a mac. Excellent. The debugger is really nice, reading .map files is very useful.

Anyway, I'm delighted to say that the problem isn't in z88dk library code, isn't in startup code, but is in fact in the music_manager code in the game - there's 3 outi in a row that emulicious complains about.

Disabling the music allows the game to run with VDP constraints enabled.  Quite why that causes so many problems - the code ends up constantly jumping to 0 if music is enabled is a bit of a mystery to someone like me with no VDP knowledge.

suborb, 2017-10-29:
Of course, the music player uses fixed addresses from $C100 to $C2xx. This overwrites tile data and
seems to cause all the weirdness that was seen.

So, as a simple fix create a buffer to provide some padding and ensure that those addresses aren't being reused for data, smc etc

commit ff9e761b (2017-04-24): Fix #192: Graphics corruption due to a misplaced ix save for
sdcc compatibility.
commit 44aa9fb1 (2017-04-25): Fix #192: Graphics corruption due to a misplaced ix save for
sdcc compatibility.
closed 2017-10-29 by suborb