z88dk/z88dk #200: sccz80: function call being generated without leading underscore, causing linking error

aralbrec, 2017-04-27:
I couldn't isolate the problem out of the source code so I've committed the example project "3D City" in z88dk/examples/sms/3dcity

The problem is in "3dcity/screen/ready_screen.c".

Translate to asm; it doesn't matter what the optimization level is:
`zcc +sms -vn -a ready_screen.c`

In the output file "ready_screen.c.asm" (which I have also committed) you will see a function call to a function missing the leading underscore at about line 47:

```
	ld	l,(hl)
	ld	h,0
	push	hl
	call	engine_tile_manager_load  ;;;;**** missing leading underscore
	pop	bc
	call	_engine_tile_manager_draw
```

The header for this function is "3dcity/engine/tile_manager.h".

If you change the prototype `void engine_tile_manager_load(unsigned char level);` to `void engine_tile_manager_load();` the leading underscore will reappear.

suborb, 2017-04-28:
This is an interesting one, I'm surprised it's not come up before with other ports. It's caused by included <sms.h> after the user headers.

The first function in sms.h is (I added it a couple of days ago), prototyped as (). This follows a path which ends up setting the __LIB__ path on the previously defined function with a proper ANSI style prototype.

I've got a fix for it so I'll drop it in this evening.

aralbrec, 2017-04-28:
I'm also missing some warnings from sccz80 when the user is not providing a prototype for functions and the compiler is assuming the prototype.  And warnings when the user calls functions of this type when the number of parameters do not match.  sdcc is calling the latter errors but I don't think sccz80 can do that given how function pointer calls are done.

3d city was loaded with these problems, missing a lot of function prototypes that sccz80 was letting slide.  I had to fix up everything to make it work with zsdcc.  It's very error prone for the user too.

suborb, 2017-04-28:
> I'm also missing some warnings from sccz80 when the user is not providing a prototype for functions and the compiler is assuming the prototype. And warnings when the user calls functions of this type when the number of parameters do not match.

It's the K&R legacy. A function prototype declared with () can have any number of arguments when defined. sdcc goes strict and (I think) assumes it's a (void) which breaks a lot of old code. Maybe we can have a warning for this one.

There's also the implicit int thing as well: any function not declared but called is assumed to return an int. We can put a warning in for that one without any consequences.

aralbrec, 2017-04-28:
> sdcc goes strict and (I think) assumes it's a (void) which breaks a lot of old code. Maybe we can have a warning for this one.

Yes sdcc cannot do K&R and will error that sort of thing.

> There's also the implicit int thing as well: any function not declared but called is assumed to return an int. We can put a warning in for that one without any consequences.

That would be good but the particular case is no function prototype at all:

```
void main(void)
{
	engine_font_manager(0,1,2);
}
```

This one will translate to asm just fine:
zcc +zx -vn -a test.c

I don't think K&R says anything about this so I think this is an error.

suborb, 2017-04-28:
> That would be good but the particular case is no function prototype at all:

> I don't think K&R says anything about this so I think this is an error. There might be some connection in here to how functions are called through function pointers currently.

Clang reports this:  `warning: implicit declaration of function 'bob' is invalid in C99` 

I added a warning for it d7d2831

If you're interested, function pointers enter callfunction() without a SYMBOL *ptr so there's no parameter information available. Functions declared as (), enters call function() with a ptr->prototyped == 0 so no parameters are checked.

Redoing the decl parsing will allow most of this stuff to be sorted out and also allow multiple levels of redirection. At the moment, there's a (int) ptr->more which points to a dummy_sym symbol that contains the dereferenced type. I think I really want a (SYMBOL *)ptr->deref which can then be followed for multiple levels of dereferencing.

suborb, 2017-04-29:
Closing this one since the underlying issue should be resolved.

closed 2017-04-29 by suborb