z88dk/z88dk #205: z88dk: improve access to block i/o instructions

aralbrec, 2017-04-28:
At the moment the new c lib contains the following sort of functions to access the zilog family's block i/o.  The classic lib has something similar.
~~~
extern uint8_t z180_inp(uint16_t port);
extern void *z180_inir(void *dst,uint8_t port,uint8_t num);
extern void *z180_indr(void *dst,uint8_t port,uint8_t num);
extern void z180_outp(uint16_t port,uint8_t data);
extern void *z180_otir(void *src,uint8_t port,uint8_t num);
extern void *z180_otdr(void *src,uint8_t port,uint8_t num);
extern void *z180_otimr(void *dst,uint8_t port,uint8_t num);
extern void *z180_otdmr(void *src,uint8_t port,uint8_t num);
~~~
The implementations themselves are very brief and simple.  For example, "z180_otdmr" is just "otdmr; ret".

These things can be inlined in post-processing either by copt or sdcc peephole rule.

aralbrec, 2017-04-28:
Maybe we should add regular block memory instructions too (ldir, lddr, cpir, cpdr)?  The builtins kind of do this already for memcpy, memset and so on.  This would be a second avenue to inline these instructions but not quite as good if the function call overhead can't be optimized away.

suborb, 2017-04-28:
For reference, in order for sccz80 to inline a call the following conditions must be met:

* memset(b,c,len) - at least len needs to be const
* memcpy(dst,src,len) - at least len needs to be const
* strcpy(dst,src) - always inlined
* strchr(ptr, c) - c must be const

No stack will be used for strcpy/strchr and at most one push/pop for memset/memcpy