z88dk/z88dk #206: classic: Amstrad NC instabilities

zx70, 2017-04-28:
While working (via emulation) on the nc100 and nc200 I noticed few stability problems.

The ZCN support (thus CP/M target mode), in the current z88dk development status, needs anew extra NOP instruction just before the BSSinit call, otherwise programs will crash.  Other CP/M environments don't need it.  I think an extra byte is worth the fix, but it is quite a mystery,  I already excluded many possible causes (file size, optimizer, compilation engine.. even ZCN seems a robust thing but I still suspect it).

Math lib crashes (checked only on emulator).  Micromania crashes too but only in native mode. ZCN seems more robust, in this case.   It can be also partially caused by a crappie z80 emulation, not easy to spot.

suborb, 2017-04-28:
Does it work better with -lmath48? Genmath uses undocumented opcodes, but I think math48 is free of them.

zx70, 2017-04-29:
Good hint, no luck :(

EtchedPixels, 2017-05-01:
Have you tried the MESS/MAME emulator ?

zx70, 2017-05-02:
Yes, for what I could observe I get identical responses.
I'm excluding problems on the stack because othello runs well.
 ..not under ZCN, but there must be a different problem there, the workaround I mentioned above makes things slightly better but doesn't solve.

In native mode I could run Othello and Dstar.  Microman and gets slightly corrupted in native ROM mode and seems to work a little better with ZCN (after hacking CPM_CRT0 to make it start).   Same for all the maths based programs and many others.

EtchedPixels, 2017-05-02:
Only thing I can find from the Amstrad "documentation"

 general notes:
     most routines return carry set if successful
     unless otherwise stated assume AF corrupted, other regs preserved
     "all registers preserved" includes flags, but NOT alternate registers
     the ALTERNATE register contents can NEVER be assumed to be preserved
     (they are used as scratch registers in time critical routines)

aralbrec, 2017-05-02:
32-bit math and above uses the exx set and so does the float math.  A few other things might as well.
The same thing as is done with the cpc464 target may have to be done and that is make sure system calls and any interrupts go through an interposer that saves these registers.

zx70, 2017-05-03:
Thankyou Alan for the doc abstract  :)
+I'm using AF' sometimes in the mono graphics library.     Dealing with just a firmware call interposer wouldn't be a problem, but I already tried to disable interrupts while coswave.c or mandel.c were running and still it doesn't work right.
I'll try to trap the interrupt entry and see what happens  :)

zx70, 2017-05-05:
Tried excluding/patching both the standard interrupt entry and NMI but the problems persist.
(I know that I really blocked the interrupts because the keyboard was disabled, then).

The ZCN mode seems to be unstable due to something into "crt0_runtime_selection.asm", I'm still investigating, but the following dirty trick seems to work  🕵️ 

        defm  	"Small C+ CP/M"

        ;INCLUDE "crt0_runtime_selection.asm"

		PUBLIC		fputc_cons
	PUBLIC		_fputc_cons
	EXTERN		fputc_cons_native
	defc fputc_cons = fputc_cons_native
	defc _fputc_cons = fputc_cons_native