z88dk/z88dk #208: Makefile issue

aralbrec, 2017-04-28:
This is just driving me nuts.

Here's a portion of the makefile I am trying to use to build the new c library:
```
${jobs}: job-%:
	@if [ "$(target)" = "z180" ]; then \
		eval $(eval cpu = --cpu=z180); \
	else \
		eval $(eval cpu = ); \
	fi;
	m4 -DCFG_ASM_DEF target/$(target)/config.m4 > target/$(target)/config_$(target)_private.inc
	m4 -DCFG_ASM_PUB target/$(target)/config.m4 > target/$(target)/config_$(target)_public.inc
	m4 -DCFG_C_DEF target/$(target)/config.m4 > target/$(target)/config_$(target).h
	cp target/$(target)/config_$(target)_private.inc config_private.inc
	@if [ "$(target)" = "zx" ]; then \
		zcc +z80 -vn -clib=new --no-crt -g -Ca"-DSTRIPVECTOR" arch/zx/bifrost2/z80/BIFROST2_ENGINE.asm.m4 -o arch/zx/bifrost2/z80/bifrost2_engine_48.bin; \
		zcc +z80 -vn -clib=new --no-crt -g -Ca"-DPLUS3 -DSTRIPVECTOR" arch/zx/bifrost2/z80/BIFROST2_ENGINE.asm.m4 -o arch/zx/bifrost2/z80/bifrost2_engine_p3.bin; \
		zx7 -f arch/zx/bifrost2/z80/bifrost2_engine_48.bin; \
		zx7 -f arch/zx/bifrost2/z80/bifrost2_engine_p3.bin; \
	fi;
	@if [ -f target/$(target)/library/$(target)_macro.lst ]; then \
		zcc +z80 -vn -clib=new -m4 --lstcwd @target/$(target)/library/$(target)_macro.lst; \
	fi;
	z80asm $(cpu) -xlib/sccz80/$(target) @target/$(target)/library/$(target)_sccz80.lst
	z80asm $(cpu) -xlib/sdcc_ix/$(target) -D__SDCC -D__SDCC_IX @target/$(target)/library/$(target)_sdcc_ix.lst
	z80asm $(cpu) --IXIY -xlib/sdcc_iy/$(target) -D__SDCC -D__SDCC_IY @target/$(target)/library/$(target)_sdcc_iy.lst
```

What is new here is I am trying to define a "cpu" variable to control when z80asm needs to assemble z180 code.  Initially I tried to set "cpu" to nothing and execute an "if" to set it to z180 but that didn't work.  So I tried this if..else thing.  The problem is both the if and else branch execute with the result that cpu is nothing.  You can see that both execute by adding echos in both branches.

I'm no expert on this... what am I doing that is stupid?

suborb, 2017-04-28:
I'm playing around with this, it should be possible.

It might be worth while using the new feature in z80asm to expand environment variables in .lst files?

suborb, 2017-04-28:
Here you go, an extract but hopefully makes sense:

```
all:
        $(eval cpu := $(shell \
        if [ "$(target)" = "z180" ]; then \
                echo "-cpu=z180"; \
        else \
                echo "";  \
        fi) )
        @echo "$(cpu)"
```

aralbrec, 2017-04-28:
Yep that did it cheers.

> It might be worth while using the new feature in z80asm to expand environment variables in .lst files?

There is likely to be a big change in how things are done in the future.  Mixing zsdcc & sccz80 objects, sorting out the peepholes into logically separate and combinable units, sharing source between classic and new, making all the pragmas compile-time (as opposed to most library build time), etc.  I'm just leaving that to the future for the time being.

closed 2017-04-28 by aralbrec

pauloscustodio, 2017-04-28:
Maybe a --cpu=z80 option can be added to z80asm? This is very easy as it does nothing.

aralbrec, 2017-04-28:
It wouldn't hurt for orthogonality's sake.