z88dk/z88dk #210: git sandbox dirty after build

vext01, 2017-04-29:
From a clean source tree, after building two files are marked modified according to git:

```
arrakis> git status
On branch openbsd
Your branch is up-to-date with 'origin/openbsd'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   lib/config/zcc.cfg
        modified:   src/config.h
```

If these files are auto-generated, would it not be best to not include them in git?

aralbrec, 2017-04-29:
Currently src/config.h is committed for the windows build using VS2015.  This is because the VS2015 build is not automatically making it so it acts as a default for windows.  Maybe we can look into that.

I'm not sure we need lib/config/zcc.cfg.  What it's there for is to select a default z80 target if one is not supplied on the compile line ("zcc +zx ...", "zcc +msx ...", "zcc +cpm ....", "zcc ..." = default z80 build currently the z88 via zcc.cfg).  Maybe we should just error out if there is no z80 target supplied?

suborb, 2017-04-29:
> On 29 Apr 2017, at 14:03, aralbrec <notifications@github.com> wrote:
> 
> Currently src/config.h is committed for the windows build using VS2015. This is because the VS2015 build is not automatically making it so it acts as a default for windows. Maybe we can look into that.
> 
That can probably go into win32/ and the include paths of the projects updated to search there as well. 
> I'm not sure we need lib/config/zcc.cfg. What it's there for is to select a default z80 target if one is not supplied on the compile line ("zcc +zx ...", "zcc +msx ...", "zcc +cpm ....", "zcc ..." = [...]
Agreed.

vext01, 2017-04-29:
Thanks! I've now submitted all of my bug reports. Sorry to rock the boat :)

suborb, 2017-04-29:
No worries. They are all valid and look easy to sort out.

vext01, 2017-04-29:
Do you guys hang in an IRC channel btw?

aralbrec, 2017-04-29:
Not unless it's being kept secret from me....

suborb, 2017-05-01:
zcc.cfg went in via #218.

As mentioned above, config.h for win32 should move to the win32 directory.

commit c47efa99 (2017-05-02): Fix #210: move vs2015 config.h into z88dk/win32
closed 2017-05-02 by aralbrec