z88dk/z88dk #211: Default library path discrepancy.

vext01, 2017-04-29:
Hi,

I'm working on packaging z88dk for OpenBSD. It builds, and `make install` puts stuff under`/usr/local` as I would expect.

It seems that zcc has a notion of a default library path, meaning that, in theory users wouldn't have to export a `ZCCCFG`. But there seems to be a discrepency between the default install prefix and the runtime path used by zcc. I first noticed this when running tests:

```
gmake -C test
gmake[1]: Entering directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test'
gmake -C machine all
gmake[2]: Entering directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test/machine'
gmake[2]: Nothing to be done for 'all'.
gmake[2]: Leaving directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test/machine'
gmake -C suites all
gmake[2]: Entering directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test/suites'
gmake -C string all
gmake[3]: Entering directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test/suites/string'
zcc +test -vn -I../../framework -Wn9 -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c stricmp.c strlcat.c strlcpy.c strrev.c strrstr.c main.c strncmp.c strcmp.c strstr.c  -o test.bin -lndos
Can't open config file /usr/local/lib/z88dk//lib/config/test.cfg
gmake[3]: *** [Makefile:15: test.bin] Error 1
gmake[3]: Leaving directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test/suites/string'
gmake[2]: *** [Makefile:24: string] Error 2
gmake[2]: Leaving directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test/suites'
gmake[1]: *** [Makefile:24: suites] Error 2
gmake[1]: Leaving directory '/usr/local/pobj/z88dk-src-1.99B/z88dk/test'
gmake: *** [Makefile:106: test] Error 2
```

Here we see zcc looking for `/usr/local/lib/z88dk//lib/config/test.cfg`, but the default install location for this file seems to be `/usr/local/share/z88dk/lib/config/test.cfg`.

(I have no `ZCCCFG` set.

aralbrec, 2017-04-29:
> I'm working on packaging z88dk for OpenBSD. 

Nice to hear!  Just a couple of points besides the issue you're raising here:

* There is a 1.99C coming in the not too distant future, fyi :)
* There is a second compiler based on sdcc called zsdcc in the project.  But this compiler's source is built separately from the rest of the z88dk binaries because we aren't holding the source in this repository.  Instead, it is currently built via [patch to the sdcc svn repository](https://www.z88dk.org/wiki/doku.php?id=temp:front#sdcc1).  When I say currently, I mean the macosx build is built this way and we have linux users build similarly using those instructions.  However, the zsdcc source is also available here on [github in the default zsdcc branch](https://github.com/z88dk/zsdcc).  I am using this to build the windows binary with VS2015 and it should be usable to build for BSD or other targets with the help of [sdcc's compile instructions in its manual](http://sdcc.sourceforge.net/doc/sdccman.pdf).  For z88dk we only need the "sdcc" and "sdcpp" binaries renamed to "zsdcc" and "zsdcpp" and placed in z88dk/bin.  We also only need "zsdcc" to be built to target the z80 family (that is, z80/z180/rabbit/tlsc90) but I haven't reduced the scope in the zsdcc repository above which should be doable with appropriate config to sdcc.  The complication is we don't have a base source for 1.99B which is what you are working with now.  Changes have been made and the current zsdcc will only work cleanly with the current z88dk codebase.  As I said, 1.99C is coming soon so then it will be possible to build zsdcc synchronized with the rest of the z88dk project.  There is another C compiler in z88dk which is being built as part of z88dk called sccz80 so all is not lost.
* There is one problem with system installs that hasn't been resolved yet.  The new c library in z88dk/libsrc/_DEVELOPMENT is intended to be easily configured by the user to change options and then have the z80 library(-ies) rebuilt.  The problem with a system install is that configuration is no longer possible without admin rights.  The correct way to resolve this is to have an optional portion of the z88dk library created in the user's home directory and have that searched before going to the system install but this has not been done yet.  By doing a system install, you are effectively freezing the library settings at their defaults.  It's not a disaster, just an impediment for more advanced users.  They may prefer to install into their home directories for this reason.

vext01, 2017-04-29:
I think I can wait for the next release tbh. Thanks.

vext01, 2017-04-30:
About the external sdcc sources. Have you considered using a git submodule?

aralbrec, 2017-04-30:
No but it looks like it might be the way to go.  I'll add it to the todo list.

suborb, 2017-05-01:
The install paths were badly mangled, I think I must've incorporated badly a debian patch several years ago. #218 makes it all consistent.

closed 2017-05-01 by suborb