z88dk/z88dk #212: Adding CFLAGS

vext01, 2017-04-29:
Hi,

I need a way to add some CFLAGS to the build. For OpenBSD ports, all C/C++ compilers should be invoked with `-O2 -pipe`. This is the default in BSD make, but curiously, not for GNU make:

```
arrakis> cat Makefile 
all:
        echo ${CFLAGS}
arrakis> make
echo -O2 -pipe 
-O2 -pipe
arrakis> gmake
echo 

```

Reading `build.sh` it looks like there is a known bug to do with this:
```
#CFLAGS="-g -O2"	# "make -e" below overrides CFLAGS in the called Makefiles by this one
					# causing problems building z80asm
```

Indeed, setting `CFLAGS` breaks the build:

```
arrakis> CFLAGS="-O2 -pipe" ./build.sh
echo '#define PREFIX "/usr/local/lib/z88dk"' > src/config.h
echo '#define UNIX 1' >> src/config.h
echo '#define Z88DK_VERSION "20170429-b7777da63"' >> src/config.h
gmake -C src/appmake
gmake[1]: Entering directory '/home/edd/source/z88dk/src/appmake'
gcc -O2 -pipe  -c appmake.c
gcc -O2 -pipe  -c z88.c
gcc -O2 -pipe  -c zxvgs.c
gcc -O2 -pipe  -c zx.c
gcc -O2 -pipe  -c z88shell.c
gcc -O2 -pipe  -c abc80.c
gcc -O2 -pipe  -c zx81.c
gcc -O2 -pipe  -c msx.c
gcc -O2 -pipe  -c mtx.c
gcc -O2 -pipe  -c mz.c
gcc -O2 -pipe  -c nec.c
gcc -O2 -pipe  -c p2000.c
gcc -O2 -pipe  -c px.c
gcc -O2 -pipe  -c aquarius.c
gcc -O2 -pipe  -c c7420.c
gcc -O2 -pipe  -c rom.c
gcc -O2 -pipe  -c sorcerer.c
gcc -O2 -pipe  -c sos.c
gcc -O2 -pipe  -c svi.c
gcc -O2 -pipe  -c sc3000.c
gcc -O2 -pipe  -c ace-tap.c
gcc -O2 -pipe  -c hex.c
gcc -O2 -pipe  -c lynx.c
gcc -O2 -pipe  -c rex6000.c
gcc -O2 -pipe  -c tixx.c
gcc -O2 -pipe  -c nascom.c
gcc -O2 -pipe  -c cpc.c
gcc -O2 -pipe  -c m5.c
gcc -O2 -pipe  -c mc.c
gcc -O2 -pipe  -c newbrain.c
gcc -O2 -pipe  -c newext.c
gcc -O2 -pipe  -c sms.c
gcc -O2 -pipe  -c trs80.c
gcc -O2 -pipe  -c c128.c
gcc -O2 -pipe  -c galaksija.c
gcc -O2 -pipe  -c vz.c
gcc -O2 -pipe  -c enterprise.c
gcc -O2 -pipe  -c x07.c
gcc -O2 -pipe  -c residos.c
gcc -O2 -pipe  -c inject.c
gcc -O2 -pipe  -c vg5k.c
gcc -O2 -pipe  -c z1013.c
gcc -O2 -pipe  -c extract.c
gcc -O2 -pipe  -c z9001.c
gcc -O2 -pipe  -c kc.c
gcc -o appmake  appmake.o z88.o zxvgs.o zx.o z88shell.o abc80.o zx81.o msx.o mtx.o mz.o nec.o p2000.o px.o aquarius.o c7420.o rom.o sorcerer.o sos.o svi.o sc3000.o ace-tap.o hex.o lynx.o rex6000.o tixx.o nascom.o cpc.o m5.o mc.o newbrain.o newext.o sms.o trs80.o c128.o galaksija.o vz.o enterprise.o x07.o residos.o inject.o vg5k.o z1013.o extract.o z9001.o kc.o
appmake.o: In function `get_temporary_filename':
appmake.c:(.text+0x7c2): warning: warning: strcpy() is almost always misused, please use strlcpy()
appmake.o: In function `parameter_search':
appmake.c:(.text+0xd7c): warning: warning: strcat() is almost always misused, please use strlcat()
appmake.c:(.text+0xe25): warning: warning: sprintf() is often misused, please use snprintf()
gmake[1]: Leaving directory '/home/edd/source/z88dk/src/appmake'
gmake -C src/appmake PREFIX=`pwd` install
gmake[1]: Entering directory '/home/edd/source/z88dk/src/appmake'
install appmake /home/edd/source/z88dk/bin
gmake[1]: Leaving directory '/home/edd/source/z88dk/src/appmake'
gmake -C src/copt
gmake[1]: Entering directory '/home/edd/source/z88dk/src/copt'
gcc -c -o copt.o -O2 -pipe -I. copt.c
gcc -c -o regex/regcomp.o -O2 -pipe -I. regex/regcomp.c
regex/regcomp.c: In function 'regcomp':
regex/regcomp.c:154: warning: array size (1) smaller than bound length (256)
regex/regcomp.c:154: warning: array size (1) smaller than bound length (256)
gcc -c -o regex/regerror.o -O2 -pipe -I. regex/regerror.c
regex/regerror.c:49: error: 'REG_OKAY' undeclared here (not in a function)
gmake[1]: *** [Makefile:18: regex/regerror.o] Error 1
gmake[1]: Leaving directory '/home/edd/source/z88dk/src/copt'
gmake: *** [Makefile:33: copt] Error 2
```

We see that it worked OK for appmake, but not for copt, where (I guess) my `CFLAGS` have overridden some required flags.

I've seem this kind of thing before, and always wondered what the best solution is. I'd be tempted to have the `CFLAGS` env *appended* to and flags which are absolutely necessary?

I can work around this with local patching, but I thought it would be best to raise here too.

suborb, 2017-04-29:
Hmmm, that should have worked, I think CFLAGS += is always used rather than a CFLAGS = though the copt compile has lost the -DLOCAL_REGEX which is needed.

vext01, 2017-04-29:
Interestingly, when invoked under the ports tree, `-O2 -pipe` is present.

Do you want me to leave this open, so you can fix the issue mentioned in the comment in the Makefile?

suborb, 2017-05-01:
I think this one was satisfied by #218, cflags is now passed down and works on all compiles

closed 2017-05-01 by suborb