z88dk/z88dk #213: gcc hardcoded

vext01, 2017-04-29:
```
arrakis> CC=clang ./build.sh
echo '#define PREFIX "/usr/local/lib/z88dk"' > src/config.h
echo '#define UNIX 1' >> src/config.h
echo '#define Z88DK_VERSION "20170429-b7777da63"' >> src/config.h
gmake -C src/appmake
gmake[1]: Entering directory '/home/edd/source/z88dk/src/appmake'
gcc   -c appmake.c
gcc   -c z88.c
gcc   -c zxvgs.c
gcc   -c zx.c
gcc   -c z88shell.c
gcc   -c abc80.c
...
```

This is due to `CC=gcc` in `build.sh`.

I do wonder why `build.sh` exists. Could you not push these few pieces of logic down into the `Makefile`, then we can override variable in the standard way.

For example, you would have in your Makefile `CC?=gcc`, then if I wanted to use clang, I would invoke `make CC=clang`.

E.g.:

```
arrakis> cat Makefile  
CC=gcc
all:
        echo ${CC}
arrakis> gmake         
echo gcc
gcc
arrakis> gmake CC=clang
echo clang
clang
```

suborb, 2017-04-29:
> On 29 Apr 2017, at 14:12, Edd Barrett <notifications@github.com> wrote:
> I do wonder why build.sh exists. Could you not push these few pieces of logic down into the Makefile, then we can override variable in the standard way.
> 
Build.sh is just a convenience thing for a local install. 

The nightly and ci builds just invoke make with CC set so any compiler can be used - I suggest your portfile does the same.

vext01, 2017-04-29:
OK, thanks.

Perhaps we can make this prominent in the README?

vext01, 2017-04-30:
It's actually easier to use build.sh because some of the environment is required.

suborb, 2017-05-01:
Leaving open because I don't think we have anything in the docs about how to compile from source.

suborb, 2019-01-30:
There’s a page on the wiki about building from source so I’m now closing this one.

closed 2019-01-30 by suborb