z88dk/z88dk #214: install target broken on master?

vext01, 2017-04-30:
```
...
cd src/z80asm ; gmake PREFIX=/usr/local/pobj/z88dk-20170429/fake-amd64//usr/local install
gmake[1]: Entering directory '/usr/local/pobj/z88dk-20170429/z88dk-040dfaa3cf23a249fc2562affc4d921bcc54bdb7/src/z80asm'
install z80asm /usr/local/pobj/z88dk-20170429/fake-amd64//usr/local/bin/
install asmpp.pl  /usr/local/pobj/z88dk-20170429/fake-amd64//usr/local/bin/
gmake[1]: Leaving directory '/usr/local/pobj/z88dk-20170429/z88dk-040dfaa3cf23a249fc2562affc4d921bcc54bdb7/src/z80asm'
cd src/zcc ; gmake PREFIX=/usr/local/pobj/z88dk-20170429/fake-amd64//usr/local install
gmake[1]: Entering directory '/usr/local/pobj/z88dk-20170429/z88dk-040dfaa3cf23a249fc2562affc4d921bcc54bdb7/src/zcc'
install zcc /usr/local/pobj/z88dk-20170429/fake-amd64//usr/local/bin/
gmake[1]: Leaving directory '/usr/local/pobj/z88dk-20170429/z88dk-040dfaa3cf23a249fc2562affc4d921bcc54bdb7/src/zcc'
./config.sh /usr/local/share/z88dk/ z88
gmake: ./config.sh: Command not found
gmake: *** [Makefile:98: install] Error 127
```

`config.sh` used to be in the top-level source directory, but was removed in c7c5e6ee9348680974c49198d2aa62632f85b5be. Looks like you forgot to update one reference in the Makefile?

I suspect you want something like:
```
$OpenBSD$
--- Makefile.orig	Sun Apr 30 15:57:50 2017
+++ Makefile	Sun Apr 30 15:57:57 2017
@@ -95,7 +95,7 @@ install:
 	cd src/sccz80 ; $(MAKE) PREFIX=$(DESTDIR)/$(prefix) install
 	cd src/z80asm ; $(MAKE) PREFIX=$(DESTDIR)/$(prefix) install
 	cd src/zcc ; $(MAKE) PREFIX=$(DESTDIR)/$(prefix) install
-	./config.sh $(prefix_share)/z88dk/ $(DEFAULT)
+	$(INSTALL) lib/config/$(DEFAULT).cfg $(DESTDIR)/$(prefix_share)/z88dk/lib/config/
 	cp -R -p include $(DESTDIR)/$(prefix_share)/z88dk
 	cp -R -p lib $(DESTDIR)/$(prefix_share)/z88dk
 	find $(DESTDIR)/$(prefix_share)/z88dk -type f -exec chmod 644 {} \;
```

But please check.

I've had to work around some CFLAGS issues with patching. I had to remove `-e` from a make invocation in build.sh otherwise things go wrong. I also found that `${INSTALL}` was empty, so I've had to invoke `make install INSTALL=install` :(

suborb, 2017-04-30:
I'm working on various tweaks to the build in the bugfix/openbsd_issues branch. I'd spotted the install problem, I'd be interested in your feedback on where it is at the moment.

suborb, 2017-05-01:
This one went in via #218

closed 2017-05-01 by suborb