z88dk/z88dk #217: z88dk: bring zsdcc into the z88dk/z88dk project as a git submodule

aralbrec, 2017-04-30:
Look into this so that zsdcc can be built as part of z88dk yet maintained separately.

sdcc has more dependencies so it may be more difficult for users to build successfully.  Maybe a build should logically separate it so that it can fail and its build resumed yet the rest of the project can be built without it.

We could bring sdcc in as a submodule directly and then apply our changes inside z88dk/z88dk to form zsdcc.  I'm not sure how that would work yet as there needs to be two separate branches in zsdcc... one for a clean re-implementation of our changes applied to sdcc and one for zsdcc as it is now.  The clean re-implementation is there to remove the hackery and act as the source for patches pushed upstream.  The zsdcc as it is now is what we're using to build the zsdcc binary.  Ultimately the clean re-implementation will be come zsdcc.

The other option is to use z88dk/zsdcc as the git submodule.

suborb, 2017-04-30:
At this point in time we should probably use z88dk/zsdcc as the submodule.

suborb, 2019-02-09:
Travis now checkouts, patches and build zsdcc.

Nightly builds use pre-built binaries to save cross-compilation fun.

Given that we don't make many changes, I think the way it's currently done is ok.

closed 2019-02-09 by suborb