z88dk/z88dk #221: z80asm: remove '#' operator

pauloscustodio, 2017-05-01:
The '#' operator was used to force an expression to be computed at assembly time, as the linker was
not able to postpone address expressions to the link phase.

With the current linker I suspect this is no longer needed. Check and remove the operator if so.

suborb, 2017-05-01:
The optimisation rules use # quite a lot to force a difference between effectively ld hl,NN and ld hl,(NN).

We generally fold up the expressions and stick them in parenthesis mod 65536 etc

Without the # operator we may end up in an ambiguous situation

pauloscustodio, 2017-05-01:
Can we use a unary plus instead?

```ld hl,+(24)``` is ```ld ld,NN```

suborb, 2017-05-01:
Sure, yup.

zx70, 2017-05-29:
Opt files are so delicate..

suborb, 2017-08-22:
This work is now folded into issue_221_remove_hash_constants

commit e7aa55d3 (2022-01-08): Fix #221: removed remaining references to the # constant operator
The operator was removed from z80asm in commit 7bd0d65d4bb56b4259f519bcbf81282d43c49779 (Mon Jul 22 16:18:57 2019 +0100)
closed 2022-01-08 by pauloscustodio