z88dk/z88dk #222: z80asm: Can I use the assembler to make a .tap image, like pasmo does?

pauloscustodio, 2017-05-01:
See #218 

Call appmake from within z80asm, e.g. appmake  +zx -b slowdos.bin --org  32768 -o slowdos.tap

vext01, 2017-05-01:
Yeah, that would be great. I did indeed note that when I called appmake, the address had to match my `org` statement in my asm src. Automating that would be great.

aralbrec, 2017-05-01:
I don't think this is a good idea.  See this [comment](https://github.com/z88dk/z88dk/pull/218#issuecomment-298354416).

The zx is not the only target, entering "appmake" on the command line will show how many there are that need special post-processing.  Even within the zx, there are many scenarios such as how to deal with bankswitched memory, whether the destination is rom (an if2 cart, eg) or disk, etc.

z80asm would explode in complexity if it had to deal with this which is why we've separated that problem into a separate utility in appmake.

If how things are done now is not convenient enough, I think it should be taken up in zcc.

pauloscustodio, 2017-05-01:
I agree that the way things are makes the toolchain more flexible, but it also increases the complexity for new-comers.

I think z80asm should have an option to call appmake in it's simplest form for a single block of linked memory, passing appmake the correct org and file names. Something in the line of:
```z80asm +zx test.asm```
would call:
```
z80asm -b test.asm
appmake +zx -b test.bin --org <appropriate-org-from-link> -o test1.tap
```

aralbrec, 2017-05-01:
Maybe it's possible to do the same for the other appmake targets with a very simple default appmake invoke.

There are several assemblers that do this sort of thing; it's just that they do it mainly because they have no environment or proper capability to generate binaries for varied environments.  However on some targets with other popular assemblers I agree that maybe doing this makes it easier for newcomers to use z80asm.

The --no-crt route almost does it very simply for the +zx:

zcc +zx -v --no-crt file1.asm file2.asm -o prog -m -create-app -Cz"--org 32768"

Where there must be an org somewhere in the asm files and it must agree with the org communicated to appmake via -Cz.

But you have to be consistent with the org.  Appmake for other targets may not be able to pick up the output binary without more options, not sure.

pauloscustodio, 2017-05-03:
Fixed with 7201cb567d3f341162037c2242eed38e7e5d1f53

pauloscustodio, 2017-05-03:
This enhacement is implemented. Now you can call:

- z80asm +zx file.asm... to generate a file.tap with the code in a REM
statement
- z80asm +zx -r30000 file.asm... to generate a file.tap with the code above
RAMTOP at 30000
- z80asm +zx81 file.asm... to generate a file.P with the code in a REM
statement

Please let me know of any further issues.

pauloscustodio, 2017-09-23:
appmake now requires a map file. Need to create it if +zx or +zx81 is requested.

aralbrec, 2017-09-23:
It shouldn't require a map file but I see it's demanding one.  It's skipping tap generation and looking at sna and binary.

pauloscustodio, 2017-09-23:
Remove map file requirement with +zx and +zx81

commit d5b0c334 (2017-09-23): Fix #222: add a map file if +zx or +zx81 are given
commit 1ea3bf4d (2017-09-23): Fix #222: map file is not required for +zx or +zx81

appmake was fixed, revert change in z80asm
closed 2017-09-23 by pauloscustodio