z88dk/z88dk #223: _DEVELOPMENT lib fails assmbly for z180

pauloscustodio, 2017-05-01:
In the z80asm_rabbit branch I'm getting assembly errors for the Z180 in the library. The cause is the use of lower or higher 8 bits of the IX and IY registers, which is not implemented in the Z180.

```
Error at file 'adt/b_array/c/sccz80/b_array_insert_n.asm' line 25: illegal identifier
Error at file 'adt/b_vector/c/sccz80/b_vector_insert_n.asm' line 25: illegal identifier
Error at file 'alloc/balloc/c/sccz80/balloc_addmem_callee.asm' line 22: illegal identifier
Error at file 'stdio/z80/asm_fflush_unlocked.asm' line 46: illegal identifier
Error at file 'stdio/z80/asm_fflush_unlocked.asm' line 47: illegal identifier
Error at file 'stdlib/z80/asm_lltoa.asm' line 41: illegal identifier
Error at file 'stdlib/z80/asm_lltoa.asm' line 42: illegal identifier
Error at file 'stdlib/z80/asm_ltoa.asm' line 43: illegal identifier
Error at file 'stdlib/z80/asm_ltoa.asm' line 44: illegal identifier
Error at file 'stdlib/z80/asm_ulltoa.asm' line 39: illegal identifier
Error at file 'stdlib/z80/asm_ulltoa.asm' line 40: illegal identifier
Error at file 'stdlib/z80/asm_ultoa.asm' line 40: illegal identifier
Error at file 'stdlib/z80/asm_ultoa.asm' line 41: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 77: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 124: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 126: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 134: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 140: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 197: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 207: illegal identifier
Error at file 'stdlib/z80/__strtoul__.asm' line 248: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 71: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 120: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 122: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 130: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 157: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 167: illegal identifier
Error at file 'stdlib/z80/__strtoull__.asm' line 198: illegal identifier
Error at file 'string/c/sccz80/memccpy_callee.asm' line 22: illegal identifier
Error at file 'string/z80/asm_strncasecmp.asm' line 44: illegal identifier
Error at file 'string/z80/asm_strncasecmp.asm' line 49: illegal identifier
Error at file 'string/z80/asm_strncasecmp.asm' line 69: illegal identifier
```

aralbrec, 2017-05-01:
> The cause is the use of lower or higher 8 bits of the IX and IY registers, which is not implemented in the Z180.

I think they are implemented for the z180.  Have a look at page 207 in the [z180 tech manual](https://github.com/z88dk/techdocs/blob/master/zilog/z180_cpu_um50.pdf):

```
Note: Suffixed H and L to ww, xx, yy, zz (ex. wwH, IXL) indicate upper and lower
8-bit of the 16-bit register respectively.
```

It's mentioning IXL by name.

I'll see if there's anything clearer about this or maybe we can ask Philip to verify this on a real z180 if it comes to it.

suborb, 2017-05-01:
p248 strongly suggests that they aren't implemented though. As in the dd prefix changes hl or (hl) only.

aralbrec, 2017-05-01:
No I think you're right.  IXL may be notational-only there.
In contrast, the z380 and ez80 manuals are very precise about their support for IXL/IXH/IYL/IYH.

Is that a complete list of the affected functions?  I'll start working on z180 alternatives.

pauloscustodio, 2017-05-01:
The list is the total list of errors coming up on the Travis-CI build. You can see the errors in Travis on the z80asm_rabbit branch.

closed 2017-05-02 by aralbrec

aralbrec, 2017-05-02:
@suborb "config_private.inc" now needs to be included in the strtol/strtoll functions in the new c lib so that the z180 version does not use ixl/ixh.  These functions are used in z80.lst in the classic lib but the "config_private.inc" file in libsrc/z80_crt0s is not visible when z80.lst is built.  I added another "config_private.inc" in libsrc so things will compile but you may want to fix that or at the least get this new file to include the other so that there is only one real copy being used.

commit 075152b7 (2017-05-02): Fix #223: eliminate use of ixl/ixh in library functions for the z180