z88dk/z88dk #226: nightly build: win32 compile of zcc.exe is broken

aralbrec, 2017-05-03:
There is a bug in the cross-compiled win32 zcc.exe that is not present in the vs2015 binaries.
See the [forums](https://www.z88dk.org/forum/viewtopic.php?pid=15025#p15025).

To reproduce, compile the sp1 demo in z88dk/libsrc/_DEVELOPMENT/EXAMPLES/zx/demo_sp1/demo1 with:
`zcc +zx -v -clib=new -startup=31 @sp1demo.lst -o demo -create-app`
In the messages generated by zcc you'll see this when processing the crt:
~~~
m4 -I "c:/z88dk/lib/config\..\..\\libsrc\_DEVELOPMENT\target\zx" -I "c:/z88dk/li
b/config\..\..\/src/m4"  --define=__STARTUP=18359902858641439 ....
~~~
The startup value should be 31 and not "18359902858641439".  In the vs2015 binary it is 31.

The relevant bit of code is [line 1011 in zcc.c](https://github.com/z88dk/z88dk/blob/master/src/zcc/zcc.c#L1011). 31 is scanned first with sscanf and then printed to generate the m4 command with snprintf.

It may also be worthwhile to check if the macosx binaries and a local linux install behave properly too.

aralbrec, 2017-05-03:
It still fails after this change.  There is something wrong with the cross compiler's implementation of %li in scanf.  I don't know if things can be updated?

suborb, 2017-05-03:
This looks like a 32/64 bit problem.

18359902858641439 = 0x413A3C0000001F

So the scanf is resolving 32 bits correctly which is what you'd expect for %ld on 32 bit machines. However the printf seems to be trying to print a 64 bit long rather than a 32 bit which is wrong.

Switching to ints here should work regardless.

aralbrec, 2017-05-03:
I've committed a standardized way to [scan exactly 32-bit quantities](https://github.com/z88dk/z88dk/blob/master/src/zcc/zcc.c#L1013).  Let's see if that works.

suborb, 2017-05-03:
I quickly built it, it's here: http://nightly.z88dk.org/misc/zcc.exe

Links:

  1. https://github.com/z88dk/z88dk/blob/master/src/zcc/zcc.c#L1013
  2. https://github.com/z88dk/z88dk/issues/226#issuecomment-298935869
  3. https://github.com/

aralbrec, 2017-05-03:
Thanks - it works so I will close.

closed 2017-05-03 by aralbrec