z88dk/z88dk #228: zsdcc: recent 9888 build does not apply peephole rules set by SO

aralbrec, 2017-05-03:
`zsdcc -v` will show the build you have.

The current build 9888 is not applying the peephole rules specified by -SO.  Going back to 9880 and everything works fine.

The result is we get builds like native sdcc with binaries several kb larger.  I'm not sure why 9888 is silently ignoring the rules yet.

aralbrec, 2017-05-03:
The bug is a [faulty patch at sdcc](https://sourceforge.net/p/sdcc/bugs/2600/).
I'll commit fixes shortly and then the zsdcc binaries should be rebuilt.

aralbrec, 2017-05-03:
Available at the [usual place](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_z88dk_patch.zip) with win32 binaries.
Now aligned with sdcc 9897.

closed 2017-05-03 by aralbrec

aralbrec, 2017-05-04:
@suborb could you replace the win32 zsdcc binary in the nightly when you get a chance?  It's the broken one that is in there now (9888)

suborb, 2017-05-04:
The next nightly should have it.

commit 0bcc905f (2017-05-03): Fix #228: repair bad fix at sdcc