z88dk/z88dk #231: sccz80: bug in initialization of auto structs

aralbrec, 2017-05-04:
Code:

```
int a, b;

struct sp1_Rect {
	unsigned char x;
	unsigned char y;
	unsigned char width;
	unsigned char height;
};

void draw_menu(void)
{
   unsigned char buffer[16];
   struct sp1_Rect r = { 10, 2, 30, 10 };
	
	a = b;
}
```

Translate to assembler:
`zcc +zx -vn -a zzz.c`

```
._draw_menu
	SECTION	data_compiler

._auto_draw_menu_r
	defb	10
	defb	2
	defb	30
	defb	10

	SECTION	code_compiler

	ld	hl,12	;const
	add	hl,sp
	ld	sp,hl
	ex	de,hl
	ld	hl,_auto_draw_menu_r
	ld	bc,4
	ldir
	ld	hl,65520	;const
	add	hl,sp
	ld	sp,hl
	ld	hl,(_b)
	ld	(_a),hl
	pop	bc
	pop	bc
	ret
```

It's moving upward into the stack instead of downward to initialize the struct.  But it's also doing things in the wrong order it seems.  First there's the +12 adjustment to do the rect and then there's a -16 adjustment to do the local buffer.

suborb, 2017-05-05:
There was a few sign issues, it should have moved down 16 + 4 and not up 16 -4. I think I've now fixed it and I'll sort out some tests later on this weekend.

aralbrec, 2017-05-05:
Yep that fixed it.
The [black star mini-game](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/zx/demo_sp1/BlackStar) now works with sccz80.  I still have to figure out why zsdcc won't play sound effects :-/

closed 2017-05-05 by aralbrec