z88dk/z88dk #233: z80asm: force the name of the output file to use section names with defined ORG

aralbrec, 2017-05-07:
In the newlib I am currently doing things like this:

~~~
SECTION DATA
org __crt_org_data

defb 0

section smc_driver
section smc_font
section smc_clib
...
~~~

for the CODE, DATA and BSS sections to make sure the output filenames corresponding to these sections will be "foo_CODE.bin", "foo_DATA.bin" and "foo_BSS.bin".  This means `appmake` only has to deal with one filename for each section but it also means users will know that those binaries correspond to.

The issue is each broadly defined CODE,DATA,BSS section is composed of many other sections; things like "code_compiler", "code_user", "code_stdio" and so on.   These sections could be empty, including the main ones CODE, DATA, and BSS.  If CODE,DATA,BSS are empty, z80asm will name the output file after the first non-empty minor section.  So if the first non-empty code section is "code_compiler" then z80asm will generate the file "foo_code_compiler.bin" instead of "foo_CODE.bin" despite the CODE section being the one having an ORG assigned.

The newlib memory maps sometimes insert a 0 byte at the beginning of CODE,DATA,BSS in order to guarantee the output filename.

I'd like to stop doing this and instead have z80asm name any output file after the code section with assigned ORG.

This is coming up because I need a way to create aligned data structures in the DATA and BSS sections.  Something like this would work if the main container DATA has an ORG that is appropriately aligned:

```
SECTION DATA
org __crt_org_data

section DATA_ALIGN_256
section DATA_ALIGN_128
section DATA_ALIGN_64
section DATA_ALIGN_32
section DATA_ALIGN_16
section DATA_ALIGN_8
section DATA_ALIGN_4
section DATA_ALIGN_2
...
```

The user or tools can put aligned structures into the appropriate sections.  If I put that 0 byte into DATA to control the output name, alignment is lost.  You could demand of the user to ORG the DATA section to one byte less than an aligned address but that would be a kludge.  

Is it possible to have z80asm name the output file after the section given the ORG address even if it's empty?

pauloscustodio, 2017-05-07:
I think it easy to do and will look into it.

aralbrec, 2017-05-17:
Paulo is there any chance this could be looked into in the near future?  I will be committing some z180 drivers soon that will use aligned buffers so it would be nice to get rid of this little pox at the same time.

pauloscustodio, 2017-05-17:
I've been busy... Let me have a look during the weekend.

commit a7bbaaca (2017-05-21): Fix #233 force the name of the output file to use section names with defined ORG

Sections with zero size where skipped and did not generate a new binary file.
Change skip condition to zero size and no ORG defined.
closed 2017-05-21 by pauloscustodio