z88dk/z88dk #235: Question: should z80asm be strict in parsing expressions enclosed in parenthesis?

pauloscustodio, 2017-05-11:
The z80asm parser needs to know if an expression has a pair of outer parenthesis in order to differentiate ```ld a,N``` from ```ld a,(NN)```. This is also used to raise a syntax error for the invalid ```ld b,(NN)```.

This mechanism is not applied today in other similar constructs, but I think it should. For instance, ```cp (NN)``` does not exist, and today the parentheses are ignored and the statement is interpreted as ```cp N```, instead of raising a syntax error.

In the z80asm_rabbit branch I've forced the later construct to raise a syntax error and got some errors compiling the library, in constructs like ```cp (XYZ) & 0xFF```.

I think the new form is better to force clarity in the code. I can volunteer to fix the few places where this appears in the library, by replacing the above with  ```cp +(XYZ) & 0xFF```.

Any objections?

aralbrec, 2017-05-11:
nope

pauloscustodio, 2017-05-23:
Decided on raising a warning instead of a syntax error.

zx70, 2017-05-29:
Warning looks good for retro compatibility.  The error stop would be legitimate but I recently had to adjust an old z80asm program and even if the errors were relevant I wished not to have to fix them all to match with the recent z80asm  :)

commit 5e0b1715 (2017-05-23): Fix #235: raise warning if (N) is used where N is expected

The instruction 'cp (23)' will now raise a warning, as 23 is interpreted
as an immediate value and not an indirect one.

To fix the warning, add a unary plus to the expression, i.e. 'cp +(23)'.
closed 2017-09-03 by pauloscustodio