z88dk/z88dk #239: z80asm: empty sections with org are causing empty binaries to be produced

aralbrec, 2017-05-28:
z80asm is producing zero size binary files for sections with org that are empty.

It would be preferable to not produce any binary file if the size will be zero because a lot of these sections are optional.  You can see the behaviour with any zx spectrum compile which is currently generating binary files for the 128's memory banks even if they are empty.

aralbrec, 2017-05-31:
z80asm is still producing a lot of empty binary files so the issue is not resolved.  For example, any sms compile will produce 32 empty files.

aralbrec, 2017-06-15:
@pauloscustodio 

There's a fix for this issue in branch z80asm_239 if you'd like to verify is correct.  I saw it's not straightforward to skip over generation of zero-size binaries so what I did instead is check the binary size after the file was closed.  If the file is zero size, it is removed.

It was driving me nuts to see so many empty files being generated especially in sms compiles.

pauloscustodio, 2017-06-15:
I'll have a look. Thanks.

pauloscustodio, 2017-06-18:
Merged, thanks @aralbrec .
Kept your changes and updated the tests.

commit 67e9c697 (2017-06-18): Fix #239 z80asm: empty sections with org are causing empty binaries...

... to be produced (thank you @aralbrec)

Merge changes by @aralbrec, update the tests.
commit 5d072f17 (2017-06-18): z80asm: Fix #239 Eliminate zero size output binaries
closed 2017-06-18 by pauloscustodio