z88dk/z88dk #240: classic (+zx): something is placed in the 128's bank 7 when compiling

aralbrec, 2017-05-28:
I'm not sure what is being put there but it's certainly a bug.

suborb, 2017-05-28:
I don't see anything that actually uses bank7 in the tree! What compile is causing the problem?

aralbrec, 2017-05-28:
There must be something in there not assigned to a section that z80asm is putting into the last defined section.

[connect.c](https://drive.google.com/file/d/0B6XhJJ33xpOWc21HYl9CdjRSNTg/view?usp=sharing)

Compile line at the top but get rid of the "max-allocs-per-node" otherwise the compile time is very long.  After the compile, bank07 contains 78 bytes.

aralbrec, 2017-05-28:
I found it:

__data_clib_head                = $C000 ; G 
__scanf_format_table            = $C000 ; G __scanf_format_table
__printf_format_table           = $C019 ; G __printf_format_table
__printf_format_table64         = $C03B ; G __printf_format_table64
__data_clib_tail                = $C04E ; G 
__tail                          = $C04E ; G 

There is no "data_clib" section defined in the classic memory map so it's getting appended to the last defined section.

I'll fix that and see if it works.

suborb, 2017-05-28:
data_clib wasn't listed as a section in crt0_section, I'm not sure why it ended up in bank7 - compiling with -Cl-v showed it in the right section,.

aralbrec, 2017-05-28:
I didn't try -Cl-v so I'm not sure if there's something else with it telling you the wrong section it's putting it in.  But appending an unknown section to the last defined section seems consistent with how memory maps are made.

After adding the data_clib section to the memory map, the compile looks right.  The program itself is not processing key input via fgets/sscanf so I'll check into that and close in a bit unless you want to keep investigating -Cl-v

commit aa626501 (2017-05-28): Fix #240: section "data_clib" was missing from the classic lib's memory map
closed 2017-05-28 by aralbrec