z88dk/z88dk #243: REL format linking

zx70, 2017-06-08:
In the past, with the Enrico Maria Giordano's help I was able to provide an object code converter to import library code elements from the CP/M world in z80asm linked programs, but I'm quite sure this is now outdated:   https://github.com/z88dk/z88dk/wiki/relformat

The right way to do it would be to make z80asm do it   🛠 😸

pauloscustodio, 2026-11-08:
> Paulo, I'm working on a nice but probably too pretentious toolset to manage the rel files.
> The idea is to expand the original bit-compacted structure to a special intermediate file describing the rel content in simple byte-based structure.
> The structure is (often) reversible to bits, which makes the rel files editable.
> 
> I've put the three tools in z88dk/support/rel.
> 
> I tried to add a feature to create z80asm object files out from a rel one I think I got close but it doesn't still work.  Now I'm beginning to understand that using the common code in z80asm etc.  Could be a better approach.
> 
> Will z80asm be ready to support the relocs in the object files structure soon?
> Would you like to take the three tools over and evolve then or integrate them the z80asm realm?
>  

 _Originally posted by @zx70 in [#1578](https://github.com/z88dk/z88dk/issues/1578#issuecomment-6713752577)_

pauloscustodio, 2026-11-08:
Stefano, I appreciate your contributions and we can integrate the REL file reading directly into z80asm. The z80asm2 did not start with the linker yet, so there is no impact from the new developments.

Three things have held me back from implementing this:
1. I have no experience at all with REL files; I've read the spec and put it down as too complex
2. I don't have a reproducible method to create REL files and use them in tests; fundamental to ensure there are no regressions in future evolution
3. I don't know what makes sense in mapping the REL structures to z80asm, i.e. what names should the fixed sections in REL have in z80asm

I think you are able to reply to the questions above, and we can make a plan of integrating your REL-file reading with z80asm.

zx70, 2026-11-08:
1) To be honest I was rather thinking at object imports rather than full rel support.
2) I'm using the existing library files to see how they were built. Once "unrolled" I could identify three potential sections: code, data and common area (the last one was never (rarely?) used.
My tools show a partial disassembly of the rel content showing rather clearly what's going on.
3) that's a good point.  Some symbol names is not z80asm compatible, using forbidden symbols. 

A viable option could be to leave everything aside but  taking the advantage of the z80asm libraries.

pauloscustodio, 2026-11-09:
Basing on existing libraries, what we need is:
- one or more .REL files that can be used as test source;
- a z88dk built program that uses symbols from those libraries;
- a way to test that the z88dk program works with the libraries, e.g. via ticks.

With that in place we can get your code to unpack the REL file and include it in z80asm; the linker will unpack the REL files and add the objects to the linkage. This is only read support, not the support to generate REL files.

In respect to the forbidden characters in symbol names, it is only a problem if the client code needs to use them. If they are internal to the REL file, the linker accepts any symbol name.

zx70, 2026-11-10:
Thank you, your summary is a great starting point. 
I could simply build samples containing code equivalent to the actual z88dk library.