z88dk/z88dk #244: newlib: open issues for zx/ts2068/next targets

aralbrec, 2017-06-11:
This is a list of open issues for the zx/ts2068/next targets in the newlib while the ts2068/next targets are made.

 [ ] The shear number of target configuration files for some targets is making it more difficult to understand those targets at a glance.  Look into moving the config files into a dedicated subdirectory.
 [ ] Both the timex and 128k spectrum have double buffered display modes.  The timex one has two display files at $4000 and $6000 with a port write selecting which one is active.  The 128k has two display files mapped to the same address in upper memory but in different banks with the active display bank selected by port write.  A target configuration should be used to indicate if dual mode is active and if so, for the 128k, the standard zx display file manipulators should be set to modify the screen mapped in upper memory instead of at $4000.  The timex dual mode is harder to accommodate as there has to be a variable someplace indicating which dfile to draw into and the display file manipulators need to read that byte when generating display file addresses.  But do in the same way for the 128k dual display mode by only enabling that code when configured.

+next
 [ ] Attempt to launch next programs using a 64k ram model with the basic rom switched out.  This may be complicated by entry points recognized by divmmc in the lower 8k.  If divmmc's interception of these important addresses cannot be disabled, look into generating a standard z88dk library in the bottom 8-16k that avoids these addresses and have programs link against that.
 [ ] Target configuration such as cpu speed, etc, may have to be set by the loader when a next program is loaded into memory.

zx70, 2017-06-11:
Rom based functions were online partially adapted to the ts2068.
I will, sooner or later, fix the ones to pass values between c and BASIC and similar stuff.

zx70, 2019-02-06:
my comments are obviously on the classiclib