z88dk/z88dk #246: Sharp PC-G8xx Family

zx70, 2017-06-13:
New target, at the moment only ANSI console mode in native resolution, no text attributes is supported.
To include as many models as possible I based the ANSI VT driver on the very minimal common API used by many sharp calculators (G-815, G-850, E-200, E-220, etc..).

zx70, 2017-06-14:
Basing on the MESS documentation, the possibly compatible models are:
PC-G801 series (1988) - intended as teaching material for schools, other models G802, G803, G805, G811, G813, G820, G830
PC-E200 (1988) - same as PC-G801, intended as a general purpose calculator.  The export version was named PC-E220.
PC-G815 (1992) - for schools
PC-G850 / G850S / G850V (1996→2002) - for schools

zx70, 2017-06-21:
Specific code for g850 console support is now in.  8 rows text taken from the SOS emulator. Something still needs to be fixed, if I keep the display data in the bss segment it works oddly.

zx70, 2017-06-29:
The high resolution display works only when the BSS section is commented out  :(
I suspect that the root cause is the same one making the nc100 code instable, but I'm not sure I understood how the BSS data is initialized..

aralbrec, 2017-06-29:
It's initialized in lib/crt0_section.asm. 

I can see a couple of potential bugs there...  `section code_crt_init` should appear after the bss and data sections have been initialized otherwise any code inserted there that changes variables will have no effect.  The BSS initialization should not also zero the data section because the data section may not be contiguous with the bss section yet the bss initialization begins at the start of bss and ldirs to the end of data.

zx70, 2017-09-19:
Issuing a clear screen (chr 12) to initialize the display solves the problem:

.ansi_cls
	ld A,' '
	ld b,24*6
	ld de,0
	jp $bfee

I suppose it is kicking the display someway or initializes some memory area   :/

closed 2017-09-19 by zx70

zx70, 2017-09-19:
Wrong cause described.
The mentioned code above is not used in "small font" mode !
The actual initialization loop IS kicking the display, see init_screen in f_ansi_char_850.asm.