z88dk/z88dk #247: z80asm - multiple ORG statements - question

feilipu, 2017-06-15:
I'm trying to assemble code that places instructions at specific locations, and possibly assembles code out of order. I've found that the z80asm doesn't seem to support this easily.

Am I doing it wrong?

1.  Locating code at specific locations.
I'd like to put code at 0x0066 (for example). In another assembler, I would do this.
`ORG 0066H`
But with z80asm, as it only supports one `ORG` I have to do this.
`defs 0x0066 - ASMPC`
This seems very clunky, as it needs a comment associated to make clear what the intent is.
Is there a better (clearer) way to absolutely locate code?

2. Assembling out of order.
If I'm trying to include code for various routines in a larger project by using the `INCLUDE` statement. But since only one `ORG` is allowed, it is not possible to have these routines create something at `0x0100`, then insert something at `0x0038`, then continue with something else at `0x2000` (all for example only).
What is the cleanest way to do that?

3. Single `ORG` statement per module.
Because z80asm only supports one `ORG` per module, if using `INCLUDE`ed files then these files have to a) specify the origin of any code they contain, which precludes using an `ORG` statement in the `INCLUDE`ing file, or b) not have any actual code because that code would precede the `ORG` statement provided in the `INCLUDE`ing file.

pauloscustodio, 2017-06-15:
z80asm uses the concept of separate compile and link. What you do with includes and several ORG per file to generate one monolithic source file to generate one binary in other assemblers, is done like this in z80asm:

1) Define the memory map in the first file to be linked. The memory map is a sequence of section instructions, in the order you want the code to appear in the resulting binary, with an ORG statement if that section is fixed in memory;
2) Each source file has section statements to tell the linker where to locate each block of code.
3) The linker does its magic without the need to include all the files into one. Each source file needs only to be assembled if needed.

For the example above, you can have for example:
memory_map.asm
```
section addr00
org 0
section addr08
org 8
...
section addr66
org 0x66
section code
; no org statement; code will follow
```

Then each source file can have:
```
section addr66
; some code
section code
; some more code
```

The files can be compiled and linked in one go, with the -d option to only compile a file if it changed, by
```z80asm -d -b memory_map.asm source1.asm source2.asm ...```

The output is a set of binary files that can be concatenated to get one memory image.

Please feel free to ask further questions if not clear.

feilipu, 2017-06-15:
Ok, thanks. That all makes sense.
I've done that and now I have a set of `.bin` files generated from below sections.
```
EXTERN      Z80_VECTOR_PROTO, Z80_VECTOR_SIZE

SECTION     z80_vector_rst
ORG         0x0000

SECTION     z80_vector_table_prototype
ORG         Z80_VECTOR_PROTO

SECTION     z80_vector_null_ret
ORG         Z80_VECTOR_PROTO+Z80_VECTOR_SIZE

SECTION     z80_vector_nmi
ORG         0x0066

SECTION     z80_acia_interrupt
ORG         0x0080

SECTION     z80_acia_rxa_chk
ORG         0x00F0

SECTION     z80_acia_rxa
ORG         0x0100

SECTION     z80_acia_txa
ORG         0x0130

SECTION     z80_acia_print
ORG         0x0180

SECTION     z80_hexloadr
ORG         0x0190

SECTION     z80_init
ORG         0x0240

SECTION     z80_init_strings
ORG         0x300
```
But now, I can't see how to encourage `appmake` to pull all these individual `.bin` files together to make a single HEX file, with the correct origin address for each `.bin` file.

What am I missing?

pauloscustodio, 2017-06-15:
You need to check/assert that each binary file has the correct size and concatenate the files together (```cat file1.bin file2.bin ... > out.bin``` in Linux or ```copy /b file1.bin+file2.bin+... out.bin```) in Windows.

Question for the appmake wizards: is there an easier way to do this with appmake alone?

aralbrec, 2017-06-15:
I'll just mention that it's rare to require so many ORGs.  There is a shift in thinking when moving to a linking assembler.  Mainly you leave address assignment to the linker with light guidance as to where rom is, where ram is and a few special addresses (like memory banks).  The cpu does have fixed entry points in the bottom 256 bytes and for that we usually just include and defs to pad in a single file.  I think this makes things cleaner.

For example, looking at this bit, does any of this have to have fixed ORG?  If you just get rid of all the ORGs, z80asm will pack it all together so there are no gaps.

```
SECTION     z80_acia_interrupt
ORG         0x0080

SECTION     z80_acia_rxa_chk
ORG         0x00F0

SECTION     z80_acia_rxa
ORG         0x0100

SECTION     z80_acia_txa
ORG         0x0130

SECTION     z80_acia_print
ORG         0x0180

SECTION     z80_hexloadr
ORG         0x0190

SECTION     z80_init
ORG         0x0240

SECTION     z80_init_strings
ORG         0x300
```

> Question for the appmake wizards: is there an easier way to do this with appmake alone?

Yes what feilipu is asking for here is an appmake tool that will pack all the output binary files into a single one according to section org.  We'd have to check for overlaps and I think we can get around a 64k restriction by looking for special strings in section names (the library is beginning to define extended memory banks with "BANK_XX" in their names). 

Something like this should be there but it's missing.  I'll take a look at it.

feilipu, 2017-06-15:
> For example, looking at this bit, does any of this have to have fixed ORG? If you just get rid of all the ORGs, z80asm will pack it all together so there are no gaps.

Yes, understood that that many `ORG` are not needed, and would probably (generally) mess up optimisation of storage, relocation etc. The "origin" of this situation was that every time I experimented or updated any of the functions (by adding or removing a byte) it would cause a flow on issues by moving the origin of the following (concatenated). Eg add a byte to the `acia_interrupt` routine, and I'd have to change the user documentation for the `print` and `hexloadr`. So, I thought that spreading the functions out by a few bytes of padding would add a little flexibility.

Also, this is also just part of the learning experience on z88dk.

> Yes what feilipu is asking for here is an `appmake` tool that will pack all the output binary files into a single one according to section org.

Yes. Exactly. Thanks.

Continuing from this (QUESTION)...
With the RC2014 (and YAZ180), for new noobs like me, it is useful to be able to assemble a `z80asm` program that can be launched in RAM with a `usr(x)` instruction from NASCOM Basic, at a particular origin or entry point. The result is that the program doesn't need a crt or any other initialisation code and can just rely on the RST08 (Tx), RST10 (Rx), RST18 (Rx CHK) serial I/O functions provided by Basic.

The natural extension of this would be to allow C code programs in the same way, but this would be more complex, as it would have to wrap in any z88dk library code needed for C functions.

Can we get `appmake` to package up HEX code starting from any nominated origin (eg 0x2000, or 0xE000), so that the package can be loaded with a HexLoadr (or similar) into the RC2014?

aralbrec, 2017-06-16:
> Eg add a byte to the acia_interrupt routine, and I'd have to change the user documentation for the print and hexloadr. So, I thought that spreading the functions out by a few bytes of padding would add a little flexibility.

Ok, yeah I can see it being a pain.  You can get zcc to make cherry pick a list of symbols and their values for, eg, documentation.

zcc .... -gf filter.txt

"-g" is going to generate a file of global defs containing the values of publicly exported symbols, including the values of public assembly labels.

The "f" part means you're going to supply filter file ("filter.txt" here) that will control which symbols are ignored and which are accepted.  Each line in "filter.txt" is a regular expression used to match against a symbol name; if you precede the regular expression with "+" a matching symbol will be accepted and if you precede with "-", it will be rejected.  The rules are applied top to bottom to each symbol until a match is made.  If "+-" is missing, the rule is assumed to be accepting, ie "+".  You can comment out individual regular expressions with a leading semicolon.

As a simple example for "filter.txt" you could just list the symbols you are interested in and reject the rest:

print
hexloadr
-[.]*

The result will be a .def file with those symbols defined using "defc".

If you change the option from "-gf filter.txt" to "-gpf filter.txt", the resulting .def file will also export the symbol so there is a "PUBLIC foo" along with "defc foo = ..".   The value of this is you can get other compiles to link against this so that you can, eg, produce a binary stored in rom and then compile programs linking against that rom.

> With the RC2014 (and YAZ180), for new noobs like me, it is useful to be able to assemble a z80asm program that can be launched in RAM with a usr(x) instruction from NASCOM Basic, at a particular [...]

Actually you can do this now.  If the code origin is not zero, the page zero code is not included and the program will be compiled for that address.   The hex file will be generated correctly.

The compile time variable controlling ORG is "CRT_ORG_CODE".  You can change its default value (0 for the rc2014) to something else via pragma which can be specified on the compile line, in a .c file or in a dedicated pragma file.   From compile line it would look like this:

zcc +rc2014 -vn ..... -pragma-define:CRT_ORG_CODE=40000 -create-app

(If you supply your own crt which you can do with another pragma, this will also replace the one provided by z88dk).

You want a ram model compile when storing a program in ram.  The rc2014 currently has named subtypes "rom" and "rom_zx7" to choose rom model and compressed rom model.  The ram model has no named subtype currently and can only be accessed via "-startup=0".  If you use that instead of subtype, a ram model compile results.  The difference is there is no separation between code, data and bss - it all gets put together into a single binary blob, unlike a rom model which will have independent org for the rom and ram parts.

pauloscustodio, 2017-06-16:
> Yes, understood that that many ORG are not needed, and would probably (generally) mess up optimisation of storage, relocation etc. The "origin" of this situation was that every time I experimented or updated any of the functions (by adding or removing a byte) it would cause a flow on issues by moving the origin of the following (concatenated) functions. Eg add a byte to the acia_interrupt routine, and I'd have to change the user documentation for the print and hexloadr. So, I thought that spreading the functions out by a few bytes of padding would add a little flexibility.

What I usually do in this case is start the code with a series of jump instructions as entry points to each routine. Then the code can move freely without impact in the external interface.

aralbrec, 2017-06-17:
I've committed "appmake +glue" which should do what is needed.  There is still one issue I have to work out with the classic library where it's using the unnamed section to store code into.

+glue understands multiple 64k memory banks.  Any section binary with "BANK_XX" in its name where XX = hex number will be understood to be part of memory bank XX.  The libraries are defining extended memory banks with this substring in section names.

+glue understands alignment.  Any section name (output binary or not) containing "align_dd" where dd = decimal number will be checked for address alignment with dd.  Only warnings are printed if alignment is wrong.  This corresponds to [point 2 of this post](https://github.com/z88dk/z88dk/issues/58#issuecomment-277497480) which describes one way alignment can be done with the sections defined in the newlib.

After zcc compiles a project, z80asm will link and output multiple binaries.  +glue will cause appmake to gather those binaries into a corresponding 64k memory bank with the help of org addresses generated in the map file.  It checks for binary overlap and alignment errors.

@feilipu

Maybe you can test this with your multiple section example.  I think you were assembling with z80asm directly but to use appmake easily things should be done through zcc.

Try this:

zcc +z180 -v --no-crt -clib=sdcc_iy test.asm -o test -subtype=none -create-app -Cz"+glue --ihex --clean"

clib=sdcc_iy although there is no c, this selects the library to link against.

no-crt is going to eliminate the one z88dk supplies so the only files involved in the assemble will be the ones you list.

subtype=none is clearing the appmake options so that the default +rom is not used.

-Cz"..." is choosing +glue.  The ihex option should be self-explanatory and the clean option will delete the separate binaries z80asm generated once they've been read.

The output files will be in the form "test__.bin" (main bank) and "test_XX__.bin" (bank xx).

feilipu, 2017-06-17:
I've tried to sort out the RC2014 code so that I can try to assemble with the below incantation, but despite a good deal of trying with different options I'm still getting a `z80asm` core dump.

It doesn't really matter how I try to call `zcc`. With `+rc2014` or `+z180`. With `-subtype=none` or `-startup=0`. There's always the same core dump, pointing to an error in the `rc2014_map.asm` file. Could it be because there is an `ORG 0x0000` statement in there?

``` bash
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr$ zcc +z180 -v --no-crt -clib=sdcc_iy @nascom32k.lst -o nascom32k -subtype=none -create-app -Cz"+glue --ihex --clean"

PROCESSING z80intr.asm
cat "z80intr.asm" >> "/tmp/tmpXXcwovzi.asm"
cat "/tmp/tmpXXcwovzi.asm" >> "/tmp/tmpXXcwovzi.tmp"
cat "/tmp/tmpXXcwovzi.tmp" >> "/tmp/tmpXXcwovzi.asm"
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z180"  -s   --cpu=z180  -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpXXcwovzi.asm"

PROCESSING int32k.asm
cat "int32k.asm" >> "/tmp/tmpXXe68vsY.asm"
cat "/tmp/tmpXXe68vsY.asm" >> "/tmp/tmpXXe68vsY.tmp"
cat "/tmp/tmpXXe68vsY.tmp" >> "/tmp/tmpXXe68vsY.asm"
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z180"  -s   --cpu=z180  -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpXXe68vsY.asm"

PROCESSING rc2014_map.asm
cat "rc2014_map.asm" >> "/tmp/tmpXXulYuGC.asm"
cat "/tmp/tmpXXulYuGC.asm" >> "/tmp/tmpXXulYuGC.tmp"
cat "/tmp/tmpXXulYuGC.tmp" >> "/tmp/tmpXXulYuGC.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z180"  -s   --cpu=z180  -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpXXulYuGC.asm"

z80asm --cpu=z180 -b -d  -o"nascom32k" -m -s -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z180" -iz180    "@/tmp/tmpXXNV3isE.lst"
z80asm: modlink.c:381: patch_exprs: Assertion `expr->target_name == NULL' failed.
Aborted (core dumped)
Errors in source file rc2014_map.asm:
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr$ 
```
I've [committed](https://github.com/feilipu/NASCOM_BASIC_4.7/commit/84f8be799c3d64d2c3a3736ee2241f2359111ed7) the code for RC2014, but it is easier to read it [from the repository](https://github.com/feilipu/NASCOM_BASIC_4.7/tree/master/sbc_NascomBasic32k_HexLoadr).

I've lots of uncertainty when to `INCLUDE` a file, versus when to compile it separately and use the contents via `PUBLIC` and `EXTERN`. I'd appreciate advice on this, too.

aralbrec, 2017-06-17:
[Direct link to memory map](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/rc2014_map.asm).

I can see two problems of which maybe Paulo can fix at least one.  I'm heading out shortly so I can't actually test it.

1.  If a source file is empty, z80asm tends to ignore it, including section definitions.   It shouldn't actually do that so maybe Paulo can fix this if this is a problem.  Again, it needs to be tested to verify but I think this may cause that NULL business.  If you want to try a quick test, put a "defb 0" someplace safe in your section map file.

2. ORG has to have an assemble-time value.  You have an EXTERN there to grab values only available at link time:  `EXTERN      Z80_VECTOR_PROTO, Z80_VECTOR_SIZE`.   At assemble time these will be 0.  So where you ORG with these values you are actually getting an "ORG 0".  These would have to be defined like "defc Z80_VECTOR_PROTO = ..." so that values are known at assemble time and you'd probably want to duplicate the intent of the EXTERN by using an INCLUDE to pull in those defc.

aralbrec, 2017-06-19:
I think "appmake +glue" is in a final state.  A couple of changes tonight:

 * the generated ihx file for a bank no longer includes padding bytes between individual sections. 
 * the crtfile is the basename of the map file used to look up symbols and binname is the basename of the output binaries generated by z80asm.  Since these are usually the same thing, crtfile will default to the basename so this simplifies a direct appmake invoke a little bit.
 * it can now deal with the no-name section that the classic library generates.

In a typical compile line with `-create-app`, add `-subtype=none -Cz"+glue --clean"`.  The "none" subtype will clear the options sent to appmake by the target.  The Cz string is then the only options sent to appmake by zcc.  Other options can be seen with "appmake +glue" on the command line.

@feilipu 
I ran out of time tonight but I will try assembling your code tomorrow to reproduce the errors you are seeing.

The matter of `include` versus `extern` isn't always clear.  The difference is down to when symbols are resolved.  Sometimes symbol values have to be known at assemble time so they will have to be done with `include`.  We're meeting one such example in this thread where you're doing an ORG with `extern` symbols.  This can't be done because ORG needs an assemble-time value but `extern` symbols' values are only known at link time.  On the other hand, if you define symbols as `public` and access them at link time through `extern`, the user will have an opportunity to override your values.  So you could make these values into defaults as part of the library and then the user can override those values in the library with his own definition file on the compile line.

I've often waffled on whether some symbol should be a defined constant or a link time value (public).  I'm even defining symbols both ways in the new library.  Two sets of defines are generated: a local set of defines which can be included into library functons at library build time and a set of public defines that are built as part of the library.  The library functions then get these constants built into the code but user functions can access the same values via `extern` without having to find the include file.  They can also override this value but I'm not sure if this has an advantage yet.

Keep it simple I think.  I'm not sure there is a right answer in every case.

feilipu, 2017-06-19:
I've made a [few minor changes](https://github.com/feilipu/NASCOM_BASIC_4.7/commit/718171745f88201254403faf58e178aa68c73b28) to the example code to get it to compile. But, I now have another issue with `appmake` and `z80asm`.

With the command line below, I'm always getting the parsing error noted below, and there are never any binaries produced.

```bash
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr$ zcc +rc2014 -startup=none --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -Cz"+glue --clean"

PROCESSING z80intr.asm
cat "z80intr.asm" >> "/tmp/tmpXX3UxnU8.asm"
cat "/tmp/tmpXX3UxnU8.asm" >> "/tmp/tmpXX3UxnU8.tmp"
cat "/tmp/tmpXX3UxnU8.tmp" >> "/tmp/tmpXX3UxnU8.asm"
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014"  -s    -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpXX3UxnU8.asm"

PROCESSING int32k.asm
cat "int32k.asm" >> "/tmp/tmpXXZ07LpE.asm"
cat "/tmp/tmpXXZ07LpE.asm" >> "/tmp/tmpXXZ07LpE.tmp"
cat "/tmp/tmpXXZ07LpE.tmp" >> "/tmp/tmpXXZ07LpE.asm"
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014"  -s    -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpXXZ07LpE.asm"

PROCESSING rc2014_map.asm
cat "rc2014_map.asm" >> "/tmp/tmpXXRqfZoD.asm"
cat "/tmp/tmpXXRqfZoD.asm" >> "/tmp/tmpXXRqfZoD.tmp"
cat "/tmp/tmpXXRqfZoD.tmp" >> "/tmp/tmpXXRqfZoD.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014"  -s    -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpXXRqfZoD.asm"

z80asm  -b -d  -o"int32k_demo" -m -s -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/rc2014" -irc2014    "@/tmp/tmpXXkAOC29.lst"
appmake +glue --clean +rom --ihex  -b "int32k_demo" -c "/tmp/tmpXXRqfZoD"
glue: Cannot set option of type output for chained command
Building application code failed
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr$
```
If I add [`DEFB    0`](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/rc2014_map.asm#L50) to the [`rc2014_map.asm`](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/rc2014_map.asm) file, then all the correct binaries are produced, but still the same parsing error from `appmake`.

I'm not sure it makes sense to have to "pollute" a `_map.asm` file with a real byte, to get the assembler to work. Wouldn't it be better to differentiate files with `_map.asm` (or otherwise `xxx.map` suffix, or something) as being pure section origin information? That would simplify their handling and make it clearer for users, I'd have thought.

The good news is that from eyeballing the generated byte code (side note, who'd have thought I'd be eyeballing z80 byte code 6 months ago... ;- ) ), the code generated looks correct.

However, the resultant linked `.bin` file doesn't have the `0x0300` string section in the right place. This is easy to spot when it is incorrect. So something is still off.

I'll have a go with some easy code, and add another comment with the result.

aralbrec, 2017-06-19:
`appmake +glue --clean +rom --ihex  -b "int32k_demo" -c "/tmp/tmpXXRqfZoD"`

Two different appmake invokes are being mixed up here (+glue and +rom).  The +rom is coming from the target's default configuration.  You're missing "-subtype=none" on the compile line which will clear away the default configuration for appmake:

`zcc +rc2014 -startup=none --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -subtype=none -Cz"+glue --clean"`

> I'm not sure it makes sense to have to "pollute" a _map.asm file with a real byte, to get the assembler to work. 

It doesn't - this will be a bug report to Paulo.  z80asm is ignoring a memory map if the file doesn't produce any bytes.

aralbrec, 2017-06-19:
`zcc +rc2014 -startup=none --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -subtype=none -Cz"+glue --clean"`

.. and the rc2014 is fubarred.  From the z88dk/lib/config/rc2014.cfg file:

```
SUBTYPE   none 
SUBTYPE   rom     -startup=1 -Cz"+rom --ihex"
SUBTYPE   rom_zx7 -startup=2 -Cz"+rom --ihex"
```

subtype=none is going to clear the startup value.  For now try:

"-subtype=none -startup=1"

and I'll have to find another way to get the appmake options to clear.

feilipu, 2017-06-21:
So I've tested the `appmake` update with this command line
```bash
zcc +rc2014 --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -subtype=none -startup=1 -Cz"+glue --ihex --clean"
```
and without `startup=1`
```bash
zcc +rc2014 --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -subtype=none -Cz"+glue --ihex --clean"
```
and provided there's a `DEFB 0` in the `rc2014_map.asm` file, then I can generate a correct Intel HEX file from either option.
[int32k_demo__.ihx.txt](https://github.com/z88dk/z88dk/files/1091447/int32k_demo__.ihx.txt)
:+1:

The next small wish is to understand how the `--filler` option works. I've tried `--filler` and `--filler 255` and `--filler255` and `--filler=255`, all without success.
:confused:

aralbrec, 2017-06-21:
> and provided there's a DEFB 0 in the rc2014_map.asm file, then I can generate a correct Intel HEX file from either option.

That's great!

> The next small wish is to understand how the --filler option works. I've tried --filler and --filler 255 and --filler255 and --filler=255, all without success.

The intel hex file is not being produced with any padding, instead there are just gaps.  I'm thinking about updating [appmake +hex](https://github.com/z88dk/z88dk/issues/119) for better hex file manipulation where you could add or extract binaries to/from hex files.  If the padding is in there then +hex won't be able to determine that a region of memory is empty.

The filler byte is, however, being put into the accompanying .bin file to fill gaps.  Do you need to have the filler byte in the hex file too?

feilipu, 2017-06-21:
>> The next small wish is to understand how the `--filler` option works. I've tried `--filler` and `--filler 255` and `--filler255` and `--filler=255`, all without success.

>`--filler 0xc9` will work.

Well, I didn't see that one coming... 😄  Anyway, so now I should be able to get it to work.

> Do you need to have the filler byte in the hex file too?

I think there's some precedent to have HEX files that are padded out with filler bytes. Certainly in the rc2014 community the Grant Searle style HEX files (with 0xff filler bytes) are pretty well known.

I would prefer to have the ability to set a filler byte. Either a `RSTxx` or a `HALT` instruction is useful for debugging. If I use a `HALT` instruction to trap bad jumps, I can see the address on the display where progress has stopped, and derive some wisdom from that. `0x00` would just run on to the next function and be confusing, as is 0xff` that just ends up at `RST38H`.

It is not a "must have", as I can get my programmer to fill free space with any value before loading the HEX, but more along the lines of "if it is easy, then yes please". TIA.

Another "nice to have" in `appmake`, would be to select the number of bytes encoded per line of HEX code.

That way, instead of a binary diff for checking results, it would be possible to diff outputs from alternate compilers and assemblers (with possibly different default encoded bytes per line), using text diff tools, and be able to eyeball any resulting code differences in a human understandable text format. That would option certainly be useful for me.

closed 2017-06-22 by aralbrec

aralbrec, 2017-06-22:
I have added a couple more options:  `--pad` to pad generated ihex files and `--recsize n` to set the record size to n bytes for ihex files.

All options can be seen with `appmake +glue` on the command line.

As an example, the following string sent to appmake from the compile line:

`-Cz"+glue --filler 0x76 --ihex --pad --recsize 32 --clean"`

will cause padded ihex files to be generated using a record size of 32 bytes.

Let me know if you find anything is broken or should be improved.

feilipu, 2017-06-22:
Updated the [RC2014 forum on this thread](https://groups.google.com/d/msg/rc2014-z80/cGiPYtc4O8o/ohahknw5AAAJ).

All looks great!

aralbrec, 2017-06-22:
It might be simpler to do this without create-app:

```
zcc +rc2014 --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -m
appmake +glue -b int32k_demo --ihex --clean
```

The compile line will stop at the creation of binaries and a map file with appmake picking those up to do the ihex file(s).

commit 78cdca46 (2017-06-22): Fix #247:  appmake +glue optionally generates filler byte for ihex files and allows setting of record size by user