z88dk/z88dk #252: z80asm ignores _map file if 0 bytes

feilipu, 2017-06-20:
In #247 the SECTION contents of the _map file are being ignored, if there are no bytes generated.

pauloscustodio, 2017-06-20:
@feilipu, I did not manage to reproduce the error you reported. What am I missing?
```
C:\Users\paulo\git\NASCOM_BASIC_4.7\sbc_NascomBasic32k_HexLoadr>zcc +rc2014 --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -subtype=none -Cz"+glue --clean"

PROCESSING z80intr.asm
cat "z80intr.asm" >> "/tmp/tmpLbQPAqvB.asm"
cat "/tmp/tmpLbQPAqvB.asm" >> "/tmp/tmpLbQPAqvB.tmp"
cat "/tmp/tmpLbQPAqvB.tmp" >> "/tmp/tmpLbQPAqvB.asm"
z80asm -D__SDCC -D__SDCC_IY -I"C:\Users\paulo\git\z88dk\lib\config\/../..//libsrc/_DEVELOPMENT/target/rc2014"  -s    -I"C:\Users\paulo\git\z88dk\lib\config\/../..//lib" -I"/cygdrive/c/Users/paulo/git/NASCOM_BASIC_4.7/sbc_NascomBasic32k_HexLoadr"  "/tmp/tmpLbQPAqvB.asm"
Error at file '/tmp/tmpLbQPAqvB.asm' line 3: syntax error
1 errors occurred during assembly
Errors in source file z80intr.asm:
Error at file '/tmp/tmpLbQPAqvB.asm' line 3: syntax error
                   ^ ---- ../includes/z80intr.asm
```
The file z80intr.asm contains only:
```
../includes/z80intr.asm
```

feilipu, 2017-06-21:
@pauloscustodio, the issue is that when the r[c2014_map.asm](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/rc2014_map.asm) file contains no bytes, then it is skipped over, and then the `SECTIONS` that it defines are not utilised to generate multiple segmented binaries.

If the final [`; DEFB    0`](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/rc2014_map.asm#L50) is remains commented out, then only one binary file is produced. If the `DEFB    0` is enabled, then the z80asm produces multiple binaries equivalent to the `SECTIONS` defined in the `_map.asm` file.

@aralbrec noted that this effect should be reported, so hence this issue #252.

pauloscustodio, 2017-06-21:
I am trying to reproduce the z80asm core dump you experienced. Can you give a commit ID of your repo where the core dump happens, so that I can track it down?

I tried with the current version of the repo, but the committed code does not compile, as described above. I tried replacing the ```../includes/z80intr.asm``` by ```include "../includes/z80intr.asm"```, but then I get several undefined symbols (the constants used in the ORG statements).

Thanks.

feilipu, 2017-06-21:
Hi @pauloscustodio,

with the [current committed code](https://github.com/feilipu/NASCOM_BASIC_4.7/commit/5a244b9449a41b0a5ed075356d72c6eef3e1565a) it is not a core dump that is the problem. The issue is simply that the `rc2014_map.asm` section map file is being ignored, unless there is a null byte present.

I've attached two assembly results. One (incorrect) assembled with the `; DEFB    0` commented out. And, a second (correct) result, when the [`DEFB    0` is included in the compilation](url).

The issue is easy to spot, because the strings should originate at `0x0300`, but in the incorrect assembly case, where the `rc2014_map.asm` file is being ignored because it is "empty" (no DEFB 0) null byte, there is nothing assembled at `0x0300`.

[int32k_demo__.bin.zip](https://github.com/z88dk/z88dk/files/1091393/int32k_demo__.bin.zip)

Command line to assemble the code provided in the attached file;
```bash
zcc +rc2014 --no-crt -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 @nascom32k.lst -o int32k_demo -create-app -subtype=none -Cz"+glue --clean"
```

pauloscustodio, 2017-06-21:
Thanks for the info. I'll have a look tonight and let you know.

aralbrec, 2017-06-21:
I've found the problem - z80asm is not storing section information in an object file if the file generates no bytes.

Use these two source files for testing.

map.asm -- this defines the memory map
```
section s0
org 0
section s1
section s2
section s3
section a0
org 0x100
section a1
section b0
org 0x200
section b1
```

source.asm -- places a numbered byte in each section
```
section s0
defb 0

section s1
defb 1

section s2
defb 2

section s3
defb 3

section a0
defb 10

section a1
defb 11

section b0
defb 20

section b1
defb 21
```

If z80asm is asked to assemble these together straight to binary, it's retaining memory map information so that the output is correct.

```
z80asm -b -o=test -m map.asm source.asm
appmake +glue -b test -c map --clean
```

The appmake invoke is gathering the separate binaries generated into a single output bin "test__.bin".  If you look at this file, everything is correct with section s0 at address 0, section a0 at address 0x100 and section b0 at address 0x200.

However, if you assemble each file to object first and then link them, the output is not correct:

```
z80asm map.asm source.asm
z80asm -b -o=test -m map.o source.o
appmake +glue -b test -c map --clean
```

Looking at the output binary you will see the section information in "map.asm" has been ignored and everything has been squashed together at address 0.

We can fix that by making "map.asm" non-zero in size by adding a defb:

"map.asm"
```
section s0
org 0
section s1
section s2
section s3
section a0
org 0x100
section a1
section b0
org 0x200
section b1
defb 0  ;; this file now generates a byte so output is non-empty
```

```
z80asm map.asm source.asm
z80asm -b -o=test -m map.o source.o
appmake +glue -b test -c map --clean
```

Now the output is correct.

pauloscustodio, 2017-06-21:
The bug with the zero-length object file is solved.

Is the core dump of z80asm reproducible? If yes, please open a new issue.

commit 1167cd67 (2017-06-21): Fix #252 z80asm ignores _map file if 0 bytes

The check for empty object file was using the total size written
instead of the fact that any data was written. Empty sections with
ORG statements were written to the object file, but discarded afterwards
if the total bytes written was zero.

Fixed.
closed 2017-06-21 by pauloscustodio