z88dk/z88dk #254: sccz80: extern functions improperly made public for fastcall / callee linkage and empty parameter list

aralbrec, 2017-06-21:
test.c:

```
extern void MusicDriverInit() __z88dk_fastcall;
extern void MusicDriverPlay() __z88dk_callee;

void main(void)
{
	MusicDriverInit();
 	MusicDriverPlay();
}
```

The intention was to make the parameter list 'void' but this syntax is still valid.

A translation to asm:

zcc +zx -vn -a test.c

makes the extern functions public:

```
	PUBLIC	_MusicDriverInit
	PUBLIC	_MusicDriverPlay
	PUBLIC	_main
```

when they should be extern.  This causes multiple symbol definition errors since these symbols are defined in another asm file.

suborb, 2017-06-21:
I thought I'd fixed that one recently. Obviously not. 

I'll take a look tomorrow. In the meantime just stick the void in there. I think that works as expected.

suborb, 2017-06-24:
That was unpleasant. The decl parser needs a proper rewrite to be honest.

closed 2017-06-24 by suborb