z88dk/z88dk #259: z80asm: show source code version in z80asm -v

pauloscustodio, 2017-06-23:
Clone of #258 for z80asm

Add output of ```git rev-list --count HEAD``` to version information.

aralbrec, 2017-06-26:
Four programs do not build including z80asm under vs2015.  For z80asm it looks like the prebuildevent is under the win32 debug version only and may work if you are only doing win32 debug builds.  I don't think the other builds will see this file if either z80asm doesn't build or z80asm isn't first in the build list.

I think the only solution is to introduce a dummy project whose sole purpose is to build that config file and then have all other projects depend on it.

I'm not going to try that now but I will commit config.h temporarily so that vs2015 builds can be done.  I think it's better to have something in the version string instead of nothing with vs2015 so I'm going to restore the build date, etc, in that file by default.

pauloscustodio, 2017-06-26:
I've temporarily reverted to the previous solution until a better one is implemented.

pauloscustodio, 2017-06-26:
Can we assume there is Perl installed in the windows machines used for building z88dk?
If yes, I can make a platform-independent Perl script that builds src/config.h. This script can then
be called from build.sh, from the top level Makefile, from Visual Studio, and from any module's Makefile that needs config.h.

aralbrec, 2017-06-26:
I think it's preferable to try to do this for a base win32 system (ie deprived of almost everything).  There should be some not-insane way of doing this for vs.

pauloscustodio, 2017-06-26:
There's one that always works: make a C program that builds config.h; compile it and run it as part of the build process. We can keep it together with the config.h, maybe make_config.c.

Any better ideas?

aralbrec, 2017-06-27:
I'm thinking about making a dummy project in the solution called "config" or similar with the sole purpose of making config.h using a PreBuildEvent like you had in z80asm.  Then have all the other projects depend on it (zcc, z80asm, etc) so that it is always made before each binary is built.

There are some issues in visual studio with the command line build process being different from the IDE process but I think the above will work for both.  I see that @zx70 has a vsbuild.cmd in the main directory for building from the command line using vs.

commit 4e78b065 (2017-06-24): Fix #259 show source code version in z80asm

Get the version string from ```git rev-list --count HEAD``` which tells how many
commits have been made, use this as the number output when z80asm is
called without arguments.
commit a0385c97 (2017-06-24): Fix #259 show source code version in z80asm

Get the version string from ```git rev-list --count HEAD``` which tells how many
commits have been made, use this as the number output when z80asm is
called without arguments.

Merge branch 'feature/z80asm_259'
commit 644b5cae (2017-06-25): Fix #259 show source code version in z80asm

See also #258, #210

Add a PreBuildEvent to the Visual Studio z80asm project file that touches src/config.h, creating an empty
file if it does not exist.

Add a rule to the z80asm Makefile to do the same.

z80asm defaults to an empty version string in src/config.h is empty. If src/config.h ws created by the
build scripts, then z80asm uses the Z88DK_VERSION string.

Remove win32/config.h
commit d89df3f1 (2017-06-26): Temporarily revert fix #259

Solve compile failures by reverting to the win32/config.h file.
Change the #ifdef in hist.c from WIN32 to _MSC_VER because gcc on cygwin defines WIN32.
Define the default version string with just the date; to write the compiler version would need a
sequence of #ifdefs for each compiler, as the compiler version is not standardized.
closed 2017-06-28 by pauloscustodio