z88dk/z88dk #263: examples/cube.c does not compile

pauloscustodio, 2017-06-24:
The file examples/cube.c is failing compilation, causing Travis-CI to fail the build tests.

```

zcc +sam -vn cube.c -o cube.bin -lm
Error at file '/tmp/tmpXXZHr8yu.asm' line 40: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 45: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 50: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 55: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 65: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 80: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 90: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 95: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 100: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 105: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 115: symbol 'i_4' not defined
Error at file '/tmp/tmpXXZHr8yu.asm' line 150: symbol 'i_4' not defined
12 errors occurred during assembly
```

The symbol ```i_4``` is not defined in the assembly file generated by zcc.

aralbrec, 2017-06-24:
There are more issues with sccz80 floats.  I'm posting another shortly that may be related.

suborb, 2017-06-24:
The builds were failing but I fixed that last night, the last few have run through successfully.

pauloscustodio, 2017-06-24:
Solved after 37f9cb79e40382a5982adc6561d08a5a4f1209f6

closed 2017-06-24 by pauloscustodio