z88dk/z88dk #266: syntax error from sdcc causes failure - question

feilipu, 2017-06-25:
I've been experiencing a failure, which I can't narrow down beyond this point. It affects everything.
It is probably just a simple thing, but I'm unable to progress after spending substantial time trashing and rebuilding everything.

With a simple program called `atest.c`
```c
main()
{
     return(0);
}
```
And this command line
```bash
phillip@SilverQuad:~/Z80/src/atest$ zcc +z80 -no-cleanup atest.c -o atest

PROCESSING atest.c
zsdcpp -D__Z88DK -D__EMBEDDED_Z80 -D__Z80 -D__SDCC -D__SDCC_IY  -D__SDCC -I"." -I"/home/phillip/Z80/z88dk/lib/config/../..//include/_DEVELOPMENT/sdcc"   "atest.c" "/tmp/tmpXXLz6h20.i2"
zpragma  < "/tmp/tmpXXLz6h20.i2" > "/tmp/tmpXXLz6h20.i"
zsdcc --constseg rodata_compiler   --reserve-regs-iy -mz80 --no-optsdcc-in-asm --c1mode --emit-externs  --no-c-code-in-asm   --no-peep --peep-file "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_peeph.2"  < "/tmp/tmpXXLz6h20.i" -o "/tmp/tmpXXLz6h20.opt"
copt "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.1" < "/tmp/tmpXXLz6h20.opt" > "/tmp/tmpXXLz6h20.op1"
copt "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.9" < "/tmp/tmpXXLz6h20.op1" > "/tmp/tmpXXLz6h20.opt"
copt "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.2" < "/tmp/tmpXXLz6h20.opt" > "/tmp/tmpXXLz6h20.asm"
cat "/tmp/tmpXXLz6h20.asm" >> "/tmp/tmpXXLz6h20.tmp"
cat "/tmp/tmpXXLz6h20.tmp" >> "/tmp/tmpXXLz6h20.asm"
z80asm -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z80"  -s    -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" "/tmp/tmpXXLz6h20.asm"
Error at file '/tmp/tmpXXLz6h20.asm' line 182: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 192: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 197: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 201: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 202: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 204: unbalanced control structure
Error at file '/tmp/tmpXXLz6h20.asm' line 215: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 219: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 220: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 221: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 222: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 226: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 227: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 231: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 235: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 238: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 239: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 240: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 241: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 242: syntax error
Error at file '/tmp/tmpXXLz6h20.asm' line 243: syntax error
21 errors occurred during assembly
Errors in source file atest.c:
Error at file '/tmp/tmpXXLz6h20.asm' line 182: syntax error
                   ^ ---- 	.globl _main
Error at file '/tmp/tmpXXLz6h20.asm' line 192: syntax error
                   ^ ---- 	.area _DATA
Error at file '/tmp/tmpXXLz6h20.asm' line 197: syntax error
                   ^ ---- %activate
Error at file '/tmp/tmpXXLz6h20.asm' line 201: syntax error
                   ^ ---- =
Error at file '/tmp/tmpXXLz6h20.asm' line 202: syntax error
                   ^ ---- %once
Error at file '/tmp/tmpXXLz6h20.asm' line 204: unbalanced control structure
                   ^ ---- ENDIF
Error at file '/tmp/tmpXXLz6h20.asm' line 215: syntax error
                   ^ ---- 	.area _DABS (ABS)
Error at file '/tmp/tmpXXLz6h20.asm' line 219: syntax error
                   ^ ---- 	.area _HOME
Error at file '/tmp/tmpXXLz6h20.asm' line 220: syntax error
                   ^ ---- 	.area _GSINIT
Error at file '/tmp/tmpXXLz6h20.asm' line 221: syntax error
                   ^ ---- 	.area _GSFINAL
Error at file '/tmp/tmpXXLz6h20.asm' line 222: syntax error
                   ^ ---- 	.area _GSINIT
Error at file '/tmp/tmpXXLz6h20.asm' line 226: syntax error
                   ^ ---- 	.area _HOME
Error at file '/tmp/tmpXXLz6h20.asm' line 227: syntax error
                   ^ ---- 	.area _HOME
Error at file '/tmp/tmpXXLz6h20.asm' line 231: syntax error
                   ^ ---- 	.area _CODE
Error at file '/tmp/tmpXXLz6h20.asm' line 235: syntax error
                   ^ ---- _main::
Error at file '/tmp/tmpXXLz6h20.asm' line 238: syntax error
                   ^ ---- 	.area _rodata_compiler
Error at file '/tmp/tmpXXLz6h20.asm' line 239: syntax error
                   ^ ---- 	.area _CODE
Error at file '/tmp/tmpXXLz6h20.asm' line 240: syntax error
                   ^ ---- 	.area _CODE
Error at file '/tmp/tmpXXLz6h20.asm' line 241: syntax error
                   ^ ---- 	.area _rodata_compiler
Error at file '/tmp/tmpXXLz6h20.asm' line 242: syntax error
                   ^ ---- 	.area _INITIALIZER
Error at file '/tmp/tmpXXLz6h20.asm' line 243: syntax error
                   ^ ---- 	.area _CABS (ABS)
phillip@SilverQuad:~/Z80/src/atest$ 
```
The attached error files (name with appended `.txt`) result.
[tmpXXLz6h20.i.txt](https://github.com/z88dk/z88dk/files/1100295/tmpXXLz6h20.i.txt)
[tmpXXLz6h20.asm.txt](https://github.com/z88dk/z88dk/files/1100293/tmpXXLz6h20.asm.txt)
[tmpXXLz6h20.err.txt](https://github.com/z88dk/z88dk/files/1100294/tmpXXLz6h20.err.txt)

This is just the simplest possible example. Everything else I try also doesn't work, but the error list is much longer.

Any thoughts or guidance on resolving this, welcome please.

aralbrec, 2017-06-25:
It looks like this file `/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_opt.1` may be damaged.  This one is putting that %once business in there and should be changing the asz80 directives to z80asm directives (.globl to GLOBAL etc).

feilipu, 2017-06-25:
@aralbrec thanks for the hint. I checked the `sdcc_opt.1` file again and you're exactly right.
It was the cause of the mismatch.

But my problem was slightly different. I had included [another older repository](https://github.com/BleuLlama/bleu-romtools) in my `${PATH}` statement, that generates a `copt` utility, which doesn't play nicely with `z88dk`. Of course, I had it in the `${PATH} first.
Easily found with a `which copt`, once you'd put me on the right track.

closed 2017-06-25 by feilipu