z88dk/z88dk #267: zsdcc - FATAL Compiler Internal Error in file 'gen.c'

feilipu, 2017-06-26:
In the most [recent versions](http://svn.code.sf.net/p/sdcc/code/trunk/sdcc/ChangeLog) of SDCC #9935, there seems to be a problem with how `zsdcc` is working with `z88dk`.

Using the standard demonstration program `umchess.c`. a direct call of `zsdcc` produces...
[umchess.c.txt](https://github.com/z88dk/z88dk/files/1101124/umchess.c.txt)
```bash
phillip@SilverQuad:~/Z80/src/umchess$ ~/Z80/sdcc-code/sdcc/bin/sdcc umchess.c -o umchess
umchess.c:29:19: error: no include path in which to search for stdio.h
umchess.c:22: warning 115: unknown or unsupported #pragma directive 'output CLIB_OPT_PRINTF = 0x00000400'
umchess.c:115: warning 84: 'auto' variable 'm' may be used before initialization
umchess.c:115: warning 84: 'auto' variable 'm' may be used before initialization
umchess.c:129: warning 112: function 'printf' implicit declaration
umchess.c:130: warning 112: function 'getchar' implicit declaration
umchess.c:129: error 101: too many parameters 
```
But, if called (when renamed and moved to  `~/z88dk/bin`)  from within`zcc`...
```bash
phillip@SilverQuad:~/Z80/src/umchess$ zcc +z80 umchess.c -o umchess

PROCESSING umchess.c
zsdcpp -D__Z88DK -D__EMBEDDED_Z80 -D__Z80 -D__SDCC -D__SDCC_IY  -D__SDCC -I"." -I"/home/phillip/Z80/z88dk/lib/config/../..//include/_DEVELOPMENT/sdcc"   "umchess.c" "/tmp/tmpXX9tlgQW.i2"
zpragma  < "/tmp/tmpXX9tlgQW.i2" > "/tmp/tmpXX9tlgQW.i"
zsdcc --constseg rodata_compiler   --reserve-regs-iy -mz80 --no-optsdcc-in-asm --c1mode --emit-externs  --no-c-code-in-asm   --no-peep --peep-file "/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/sdcc_peeph.2"  < "/tmp/tmpXX9tlgQW.i" -o "/tmp/tmpXX9tlgQW.opt"
umchess.c:115: warning 84: 'auto' variable 'm' may be used before initialization
umchess.c:115: warning 84: 'auto' variable 'm' may be used before initialization
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3720' : Shouldn't push DE if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3744' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
.
.
.
.
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3744' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3720' : Shouldn't push DE if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3744' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3720' : Shouldn't push DE if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3744' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3744' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
umchess.c:133: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '4579' : code generator internal error 
Contact Author with source code
phillip@SilverQuad:~/Z80/src/umchess$
```
Perhaps, it is just me, again...

aralbrec, 2017-06-26:
I can see the last few commits at sdcc are to do with the register allocator so maybe something has gone wrong.  I won't have time to check it tonight.  Maybe back off and checkout an older version of sdcc and make sure that still works?  I'm currently using 9918.

aralbrec, 2017-06-26:
It looks like Philip is trying to speed up the compiler.

feilipu, 2017-06-26:
You know the feeling, when you bite into an apple and it is bad..?

There has been another `sdcc` update overnight, and now all is back to normal.
I tested `#9918`, which works as suggested, and also the most recent version `#9938`.

I'll be staying with `#9938` for a while, since it is working fine now.
```bash
phillip@SilverQuad:~/Z80/src/startrek$ zsdcc -v

ZSDCC IS A MODIFICATION OF SDCC FOR Z88DK
Build: 3.6.7 #9938 (Linux) Jun 26 2017

sdcc website:
https://sourceforge.net/projects/sdcc/

patch details:
https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_z88dk_patch.zip

published under GNU General Public License (GPL)
```

closed 2017-06-26 by feilipu