z88dk/z88dk #269: nightly build:  win32 compile not recognizing "-std=gnu11" on z80asm build

aralbrec, 2017-06-27:
.

pauloscustodio, 2017-06-27:
C11 standard is suported in gcc-4.9. What version exists in the build machine? Could it be upgraded?

suborb, 2017-06-27:
The issue is that the macOS toolchain would need to be upgraded at the same time and it's a pain to get a working toolchain. Versions are:

```
build@z88dk-build:~$ i686-apple-darwin10-gcc -v
Using built-in specs.
Target: i686-apple-darwin10
Configured with: ../gcc-5646.1-i686-apple-darwin10/configure --target=i686-apple-darwin10 --prefix=/usr --with-sysroot=/usr/lib/apple/SDKs/MacOSX10.5.sdk --with-slibdir=/usr/i686-apple-darwin10/lib --mandir=/usr/i686-apple-darwin10/share/man --infodir=/usr/i686-apple-darwin10/share/info --libexecdir=/usr/lib --disable-nls --with-gxx-include-dir=/usr/include/c++/4.2.1 --enable-languages=c,c++,objc,obj-c++ --enable-version-specific-runtime-libs --with-as=/usr/bin/i686-apple-darwin10-as --with-ld=/usr/bin/i686-apple-darwin10-ld : (reconfigured) ../gcc-5646.1-i686-apple-darwin10/configure --target=i686-apple-darwin10 --prefix=/usr --with-sysroot=/usr/lib/apple/SDKs/MacOSX10.5.sdk --with-slibdir=/usr/i686-apple-darwin10/lib --mandir=/usr/i686-apple-darwin10/share/man --infodir=/usr/i686-apple-darwin10/share/info --libexecdir=/usr/lib --disable-nls --with-gxx-include-dir=/usr/include/c++/4.2.1 --enable-languages=c,c++,objc,obj-c++ --enable-version-specific-runtime-libs --with-as=/usr/bin/i686-apple-darwin10-as --with-ld=/usr/bin/i686-apple-darwin10-ld
Thread model: posix
gcc version 4.2.1 (Apple Inc. build 5646) (dot 1)
build@z88dk-build:~$ i586-mingw32msvc-gcc -v
Using built-in specs.
Target: i586-mingw32msvc
Configured with: /build/buildd-mingw32_4.2.1.dfsg-2-i386-16QVXY/mingw32-4.2.1.dfsg/build_dir/src/gcc-4.2.1-2-dfsg/configure -v --prefix=/usr --target=i586-mingw32msvc --enable-languages=c,c++ --enable-threads --enable-sjlj-exceptions --disable-multilib --enable-version-specific-runtime-libs
Thread model: win32
gcc version 4.2.1-sjlj (mingw32-2)
```

suborb, 2017-06-27:
I'll investigate the tools here: https://github.com/tpoechtrager for building some new tool chains.

pauloscustodio, 2017-06-27:
It was a real pain to setup .travis.yaml for C11 - it took me a lot of commits with failed builds to get finally a working configuration.

You can have a look at the current .travis.yaml at the master branch to see what is currently working. Any comments are welcome.

suborb, 2017-07-02:
@pauloscustodio thanks for rolling back the option, I'm working on rebuilding the build VM, I've now got gcc 6.2.0 for windows and clang 3.8.1 for MacOS.

I need to sort some issues with libxml2 (used for one of the support/graphics tools) and then I'll switch over.

suborb, 2017-07-05:
The build for 6/7/2017 will be from the new machine - can someone test that the binaries work as expected?

aralbrec, 2017-07-07:
The win32 package is 180MB so something wrong there.

suborb, 2017-07-07:
There was a change in argument parsing of tar so it ended up with the git tree.

aralbrec, 2017-07-07:
I braved the download (about 30 minutes :) .  The windows binaries seem to be working.

suborb, 2017-07-07:
Thanks. Hopefully the size should be fixed tonight. 

I think something might be up with my connection. Have noticed downloads being a bit slower than expected lately. Perhaps uploads have been similarly afflicted.

closed 2017-07-19 by suborb