z88dk/z88dk #270: z80asm: ignoring org -1 for sections when data is in object file separate from memory map file

aralbrec, 2017-06-28:
Test code:

zzz.asm
```
SECTION CODE
org 0

SECTION DATA
org 0x8000

SECTION BSS
org -1
```

yyy.asm
```
SECTION CODE
defb 1,2,3,4,5

SECTION DATA
defb 10,11,12,13

SECTION BSS
defs 3
```

First a compile that goes straight to binary:

`z80asm -b -o=zzz zzz.asm yyy.asm`

Everything is good, three files are generated:  zzz_CODE.bin, zzz_DATA.bin, zzz_BSS.bin

Next try doing the same but build object files first:

```
z80asm zzz.asm
z80asm yyy.asm
z80asm -b -o=zzz zzz.o yyy.o
```

Only two files are generated:  zzz_CODE.bin and zzz_DATA.bin.  The BSS section has been appended to DATA as if the "org -1" was ignored.

If you put everything into one file and go through object file, again the correct result.

aralbrec, 2017-07-01:
Paulo is there any chance you could have a look at this sooner rather than later?  I hadn't realized that this is a critical bug for all rom compiles - nothing will work that is destined for rom.

http://www.smspower.org/forums/16691-AnotherZ88DKSDCCDifference#98643

pauloscustodio, 2017-07-01:
Done. Thanks for reporting.

aralbrec, 2017-07-02:
Thanks Paulo, the example given now works but the problem still exists in compiles for a reason I haven't been able to determine yet.

aaa.c
```
// zcc +z80 -v -clib=sdcc_iy aaa.c -o aaa -m

// DATA

unsigned char data[] = "Hello";

// BSS

unsigned buffer[100];

// CODE

int main(void)
{
	return 1;
}
```

After the compile a separate bss binary is not generated.

The memory map used is [this one](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z80.inc#L97), excerpt here:

```
SECTION BSS

IF __crt_org_bss

   org __crt_org_bss

ELSE

   IF __crt_model

      org -1

   ENDIF

ENDIF

section bss_align_256
section bss_align_128
...
```

From the map file generated by the compile:

```
TAR__clib_malloc_heap_size      = $FFFFFFFF ; L z80_crt
TAR__crt_interrupt_mode         = $FFFFFFFF ; L z80_crt
TAR__crt_org_bss                = $FFFFFFFF ; L z80_crt
__clib_malloc_heap_size         = $FFFFFFFF ; L z80_crt
__crt_interrupt_mode            = $FFFFFFFF ; L z80_crt
__crt_org_bss                   = $FFFFFFFF ; L z80_crt
DEF__crt_enable_trap            = $0000 ; L z80_crt
IOCTL_RESET                     = $0000 ; G z80_crt
...
```

pauloscustodio, 2017-07-02:
I found and fixed another problem for the given C code. Please let me know of any further problems.

pauloscustodio, 2017-07-02:
This should be fixed now. Please let me know.

On a related but different topic: how do I get the zsdcpp binary? 
I downloaded the nightly build and copied the executables, but we should be able to build from source.

Thanks,
Paulo

From: aralbrec
Sent: 2 de julho de 2017 07:38
To: z88dk/z88dk
Cc: pauloscustodio; State change
Subject: Re: [z88dk/z88dk] z80asm: compiles defining org -1 for bss sectionnot generating a separate bss bin for unknown reason (#270)

Thanks Paulo, the example given now works but the problem still exists in compiles for a reason I haven't been able to determine yet.
aaa.c
// zcc +z80 -v -clib=sdcc_iy aaa.c -o aaa -m

// DATA

unsigned char data[] = "Hello";

// BSS

unsigned buffer[100];

// CODE

int main(void)
{
	return 1;
}
After the compile a separate bss binary is not generated.
The memory map used is this one, excerpt here:
SECTION BSS

IF __crt_org_bss

   org __crt_org_bss

ELSE

   IF __crt_model

      org -1

   ENDIF

ENDIF

section bss_align_256
section bss_align_128
...
From the map file generated by the compile:
TAR__clib_malloc_heap_size      = $FFFFFFFF ; L z80_crt
TAR__crt_interrupt_mode         = $FFFFFFFF ; L z80_crt
TAR__crt_org_bss                = $FFFFFFFF ; L z80_crt
__clib_malloc_heap_size         = $FFFFFFFF ; L z80_crt
__crt_interrupt_mode            = $FFFFFFFF ; L z80_crt
__crt_org_bss                   = $FFFFFFFF ; L z80_crt
DEF__crt_enable_trap            = $0000 ; L z80_crt
IOCTL_RESET                     = $0000 ; G z80_crt
...

closed 2017-07-02 by pauloscustodio

aralbrec, 2017-07-02:
Thanks Paulo.  Everything looks good.

commit e4672324 (2017-07-01): Fix #270 ignoring org -1 for sections when assembly and link in different steps

The split-section flag (org -1) was being reset in the linker if an object file did not define an origin.
Fixed.
commit f29e51f5 (2017-07-02): Fix #270 ignoring org -1 for sections when assembly and link in different steps

second bugfix
The split-section flag (org -1) was being ignored by the linker while creating the output files if the section
containing the flag had no data.
Fixed.