z88dk/z88dk #275: sccz80: special function registers not allowed to start with the letter S

aralbrec, 2017-07-03:
Yes you read that right :)

Special function registers cannot start with the letter 'S'

[code]
#include <stdlib.h>

__sfr __banked __at 254 SINCLAIR_ZX_SPECTRUM_BORDER;

int main(int argc, char *argv[])
{
        SINCLAIR_ZX_SPECTRUM_BORDER = 2;
        return 0;
}
[/code]

zcc +zx -vn -lndos -create-app --list --c-code-in-asm test.c -o test.bin

Result:
```
sccz80:"zzz.c" L:7 Error:#42:Unknown symbol: SINCLAIR_ZX_SPECTRUM_BORDER
Compilation aborted
```

Change the name to anything that doesn't start with 'S' and it works fine.

suborb, 2017-07-03:
> thread[2].>

Ah, it's a greedy matching for 'S' for signed. The same problem exists
with 'L' and 'U'.

Links:

  1. https://github.com/z88dk/z88dk/issues/275
  2. https://github.com/

aralbrec, 2017-07-04:
I tried A-H and found they worked so stopped there and decided only S was a problem.  Only a little ways to go to L and I could probably have figured it out :)

Looks like things are working.

closed 2017-07-04 by aralbrec