z88dk/z88dk #277: Relocatable jump / vector tables for z80 & z180 ROM memory models - question / enhancement

feilipu, 2017-07-05:
Extending from #50 and #276

Looking at the memory model for the [z80](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z80.inc) and [z180](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z180.inc), both seem to assume that the jump destination for the `RSTnn` instructions is fixed into the ROM of the device, for the ROM model.

If the z80 [`__crt_org_vector_table`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z80.inc#L135) is set positive, then the jump table location could be positioned in RAM, but there seems to be no mechanism to fill the jump table with the relevant vectors during the initialisation.

Also, the connection between the individual (eg [`RST08`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_page_zero_z80.inc#L14)) jumps and the allowed jump table is not apparent. Is this jump table for the `RSTnn` instructions? Or does it have another purpose?

If the z180 [`__crt_org_vector_table`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z80.inc#L138) is set positive, then an address vector table can be positioned in RAM, but initialisation of this table is [left to the user](https://github.com/z88dk/z88dk/issues/50#issuecomment-281574192). That is ok, but otherwise there is no mechanism to initialise them at boot.

__Enhancement Request__

Would it be possible to allow the z80/z180 RST vector jump table to be [prototyped](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z80intr.asm#L80) within the `SECTION DATA` and have the [vectors initialised](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/sbc_NascomBasic32k_HexLoadr/int32k.asm#L319) into RAM during the normal crt initialised data process?

This would allow z80 ROM builds to
* reassign the `RSTnn` vectors during run time,
* allow sharing (reassignment) of the `RST38` `INT0` and `NMI` vectors if hardware could share the signal.
* update systems if bugs are found post shipment of ROMs.

Similarly, the z180 vector address table could also be [prototyped](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z180intr.asm#L102) within `SECTION DATA` and be [initialised](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z180intr.asm#L80) at the same time.

Although the z180 can relocate its internal vector address table into RAM, this would allow the z180 vectors to have a similar (simple) interface to the z80 vectors. i.e. `INT1` and `INT2` would work the same as `INT0` for the user, even though they are managed by completely separate mechanisms.

Sorry if I'm not being clear. Please ask me to clarify.

aralbrec, 2017-07-05:
> Looking at the memory model for the z80 and z180, both seem to assume that the jump destination for the RSTnn instructions is fixed into the ROM of the device, for the ROM model....
> Also, the connection between the individual (eg RST08) jumps and the allowed jump table is not apparent. Is this jump table for the RSTnn instructions? Or does it have another purpose?

There is a [pragma that allows the user to specify destination JPs for the RSTs](https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#implementing_the_z80_restarts_im1_and_nmi).  By default the crt will fill them in with do-nothing code.

If you wanted to implement RST08 for example, you'd set bit 1 of CRT_ENABLE_RST:
`#pragma output CRT_ENABLE_RST = 0x02   // RST 08 implemented by user`

Then the compiler expects the user to define a function called `_z80_rst_08h` that the crt will jump to.  You can implement that function in asm or C with that name or if you want to use a different name you can use a pragma redirect to rename it:  `#pragma redirect _z80_rst_08h = foobaz`

> If the z180 __crt_org_vector_table is set positive, then an address vector table can be positioned in RAM, but initialisation of this table is left to the user. That is ok, but otherwise there is no mechanism to initialise them at boot.....
> Similarly, the z180 vector address table could also be prototyped within SECTION DATA and be initialised at the same time

The clumsily written and incomplete bit on im 2 is [here](https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#interrupt_service_routines).  This last part where you put the vector table into the DATA section and have the crt initialize it is what is missing and yes we can put that in there.

> This would allow z80 ROM builds to
>    reassign the RSTnn vectors during run time,
>    update systems if bugs are found post shipment of ROMs.

The RSTs are at a fixed address so if that's in rom that's it.  If you wanted it to be patchable you'd have to implement a RST function that loads its destination from a ram variable like:

[code]
0008:
   jp _z80_rst_08h   // permanent in rom
....
_z80_rst_08h:    // user defined routine for rst 08h
   ld hl,(_vector_rst_08h)
   jp (hl)
[/code]

Or you could put `_z80_rst_08h` into a self-modifying code section so that it's in ram while the program is running:

[code]
_z80_rst_08h:
   defb 195   // jump
_vector_rst_08h:
   defw default_rst_08h
[/code]

Then you can write the rst address directly after the jump opcode.

If the crt happens to be in ram, you can also write the rst address directly to 0x0009.

aralbrec, 2017-07-05:
> Although the z180 can relocate its internal vector address table into RAM, this would allow the z180 vectors to have a similar (simple) interface to the z80 vectors. i.e. INT1 and INT2 would work the same as INT0 for the user, even though they are managed by completely separate mechanisms.

That's the intention.  The negative values for __crt_org_vector_table (-0x80 and -0x100 IIRC) are to set up an im2 table inside the crt at address 0x80 and 0x100 respectively.  0x80 is a partial table where device ids must start at 0x80 (because I=0).

feilipu, 2017-07-05:
I've been reading and thinking on this issue tonight.
And, I'll have a go at solving this on z88dk terms over the next few days.

Probably just two files. One specific to z80. Another to add the z180 interrupts to it.
I'm not sure I can fully integrate it into the config and build system, but I'll try.

feilipu, 2017-07-06:
See #278 for a partial solution.

I'll update the z180 separately, to keep the pull manageable.

aralbrec, 2017-07-10:
A lot of discussion pertaining this issue is in #278 . 
I think changes have been made to satisfy the issues here too?

feilipu, 2017-07-10:
Pretty much. I'll see whether I can understand and create a sensible outcome for the z180 internal vectors, but I'm not sure whether they are general enough ( with 0, L, and S variants) to incorporate into the generic target. I'll probably request they be later added to the YAZ180 specific target, along with other drivers (over the next months)

aralbrec, 2017-07-11:
A library function can `INCLUDE "config_private.inc"` to gain access to the configuration information in the target's `config_z180_private.inc` (eg) file.  The z180's file contains:

```
defc __Z180 = 0x04

defc __Z180_Z80180 = 0x01
defc __Z180_Z8L180 = 0x02
defc __Z180_Z8S180 = 0x04
```

so you can test what type of z180 the library is being built for.  I think this is probably sufficient but that can't be known until code is actually written that depends on distinguishing z180 types.

feilipu, 2017-07-12:
I've added some more yaz180 work, and it is getting closer.

One thing I need for z180 is to set the `IL` register for the internal interrupts.
So, I've added a [commit for the `IL` register](dc2e24aa81995fcd049db75ce4b6d3dceef8426a).
But, I don't know whether `INCLUDE "config_private.inc"` is the correct way to do this.

aralbrec, 2017-07-13:
If `__crt_org_vector_table = 0`, the initialization of IL is moved to the crt out of page zero.  This is to ensure compiles that aren't at address 0 (ie there is a monitor or operating system present)  but do want to create an interrupt vector table, get IL initialized as well.

The initialized value of IL is controlled by the cpu configuation in config_cpu.m4.  The location of the vector table is a separate issue and is determined by `__crt_org_vector_table`.

aralbrec, 2017-07-17:
I think these issues have been satisfied now.

closed 2017-07-17 by aralbrec