z88dk/z88dk #279: classic: draw and undraw are not inverses of each other

aralbrec, 2017-07-08:
zzz.c
```
#include <stdio.h>
#include <graphics.h>

void main()
{
        int i;

        clg();

        for(i = 0; i < 24; i++)
        {
                draw(81, 6 + i, 86, 6 + i);
        }

        (void) getchar();

        for(i = 0; i < 24; i++)
        {
                undraw(81, 6 + i, 86, 6 + i);
        }

        (void) getchar();
}
```

zcc +zx -vn zzz.c -o zzz -create-app -lndos

The lines are not completely undrawn by `undraw`.

It looks like draw.asm plots the first point before calling line whereas undraw does not.

suborb, 2017-07-08:
Yup that's what it does do by the looks of things. I'm not too certain why though.

aralbrec, 2017-07-08:
I don't know either.  If the first pixel is skipped by the line draw maybe it should be fixed there or all functions need to plot the first point.  Maybe @zx70 has more insight.

zx70, 2017-07-12:
Undraw and xordraw code was left behind for a while, (in an attempt to use rom native calls, eg on amstrad).. recently I supported (tried to) the Dom classic lib sorting activity and spotted many things needing an alignment, this is yet another one.. i can fix myself.

zx70, 2017-07-13:
Fixed.   I suspect I left it like this to balance the border in the chessboard.c demo <blush>

closed 2017-07-13 by zx70