z88dk/z88dk #281: classic: Push x11 variables down into the library

suborb, 2017-07-12:
Xz88dk.h seems to define a number of variables. At the moment these end up in user code, but they should be pushed down into the library.

aralbrec, 2017-07-12:
The same problem may exist in game.h:

```
#ifndef MAKE_LIB

#ifdef __C128__
	unsigned char *joystick_type[] = {"Joystick 1", "Joystick 2", "QAOP-MN"};
	#define GAME_DEVICES 3
#endif
...
```

This also may be causing multiple symbol errors in the [OP's forum post](https://www.z88dk.org/forum/viewtopic.php?pid=15166#p15166).

suborb, 2017-07-12:
Also 7fdcf211cf1730fcbbab4f8ac3c912e964be6c3d fixes the original issue

suborb, 2017-07-12:
conio.h also has PCDOS_COLORS

aralbrec, 2017-07-13:
It might be better to put those things in a separate file so the onus is not on the user to use separate compilation and define the controlling label for one file.

games.h
```
#ifdef __C128__
#ifdef DEFINE_JOYSTICK_TYPE
	unsigned char *joystick_type[] = {"Joystick 1", "Joystick 2", "QAOP-MN"};
#endif
	#define GAME_DEVICES 3
#endif
...
```

The sense of the if-guard has been changed so you have to ask for the variable to be defined in at least one source file.  If "joystick_type" were put in a separate file, the linker would pick it up automatically.  "const" would also make sure it ends up in rodata for rom compiles.

suborb, 2017-07-13:
No. There's a library file that explicitly defines that to pull it in. The user has to do nothing.

commit 97e08eef (2017-07-12): Fix #281: Push joystick_type down into the library as well
closed 2017-07-19 by suborb