z88dk/z88dk #283: newlib +z180:  need to look at when the IL register is initialized

aralbrec, 2017-07-12:
Merge #282 discovered the necessity to initialized the IL register in the crt rather than the z180 page zero.

feilipu, 2017-07-12:
I'm still not sure I've properly understood how the `IM2` address vectors can be differentiated from the z180 internal interrupt address vectors.

My understanding issue is how should z88dk provide an automated for locating the z180 fixed internal interrupt vectors (which can only be moved by 32 bytes at a time), without them potentially conflicting with a z80 `IM2` location?

Both have their high byte controlled by `I`, but the z180 has the low byte controlled by `IL` plus the individual (fixed) offset for the specific vector. But, the `IM2` vectors can take any value (obtained by reading the bus) that the user chooses to provide.

> The five least significant `IL` bits are fixed for each specific interrupt source. By programming IL, the vector table can be relocated on 32 byte boundaries.

Speaking as a friend of the z180 in `IM1` mode, I'd prefer to have the `__crt_org_vector_table` value define the origin of the z180 internal vector address table (via `I` and `IL`), as its priority.

A specific z180 `IM2` solution is secondary for me, and in the case of the z80 platforms there's no conflict anyway.

aralbrec, 2017-07-13:
> My understanding issue is how should z88dk provide an automated for locating the z180 fixed internal interrupt vectors (which can only be moved by 32 bytes at a time), without them potentially conflicting with a z80 IM2 location?

The z180's internal devices are in fact z80 im2 devices.  There is only one interrupt vector table that is shared by the internal z180 devices and external im2 z80 devices connected to INT0.

Setting the IL register defines the vectors the internal z180 devices will use and usually z80 im2 devices have their vectors programmed at startup as well.  It's up to the programmer to define these things so there are no conflicts in the table.

So there's one table, IL is deciding where in this table the z180 internal devices' vectors will be.  Any external INT0 devices will have their vectors in the same table.  The programmer creates one table in `section interrupt_vectors` that holds vectors for everything.  This table must be properly padded to have the devices' vectors in the right place.  Since the table can be placed anywhere in memory, not just aligned to a 256 byte page (`__crt_org_vector_table` is not masked when the vector table is copied to ram), the vector table can be smaller than 256 bytes and doesn't have to start at vector 0 if vector 0 is not used.

> Speaking as a friend of the z180 in IM1 mode, I'd prefer to have the __crt_org_vector_table value define the origin of the z180 internal vector address table (via I and IL), as its priority.

That's what it does.  IL determines the vectors the z180 internal devices will use within that table.

aralbrec, 2017-07-17:
IL is initialized in the page zero code if the user doesn't ask for a vector table by setting CRT_ORG_VECTOR_TABLE to a non-zero value.  It is otherwise initialized in the crt.

closed 2017-07-17 by aralbrec