z88dk/z88dk #291: sccz80: Unexpected crash

Fabrizio-Caruso, 2017-07-19:
Hi

When compiling my game (https://github.com/Fabrizio-Caruso/ASCII-CHASE)
with sccz80 I get an unexpected crash in an innocent-looking portion of my code:
"
void handle_invincible_ghost(void)
{
	if(!invincibleGhost._status)
	{
		// Manage invincible ghost
		#if defined(DEBUG_SHOW_INVINCIBLE_GHOST_ISSUE) && defined(__SPECTRUM__)
			if(1) 
		#else
			if(...)
		#endif
		{
			invincibleGhost._status = 1;
			DRAW_INVINCIBLE_GHOST(invincibleGhost._x, invincibleGhost._y, invincibleGhost._imagePtr);
			#if defined(DEBUG_SPECTRUM_HACK) && defined(__SPECTRUM__)
				sleep(1);
			#endif
		}
		else
		{
...
		}
	}
	else
	{ 	
...
	}
}
"

The crash can be avoided by doing something just before its occurrence, e.g., insert a "sleep(1)" command avoids the bug.

#define DEBUG_SHOW_INVINCIBLE_GHOST_ISSUE makes the issue apparent at the beginning of the game
#define DEBUG_SPECTRUM_HACK "fixes" by adding a "sleep(1)" command just after the buggy code is entered 

I am compiling the code with
WITH ISSUE: zcc +zx -SO3 -DDEBUG_SHOW_INVINCIBLE_GHOST_ISSUE -vn -DAMALLOC -clib=ansi -lmalloc -lndos -create-app  -o %deliverables%\ZXSpectrum_ISSUE.prg %mypath%\display_macros.c %mypath%\powerUps.c %mypath%\enemy.c %mypath%\invincible_enemy.c %mypath%\level.c %mypath%\character.c %mypath%\text.c %mypath%\missile.c %mypath%\strategy.c %mypath%\input.c %mypath%\main.c

WITH "FIXED" ISSUE: zcc +zx -SO3 -DDEBUG_SPECTRUM_HACK -DDEBUG_SHOW_INVINCIBLE_GHOST_ISSUE -vn -DAMALLOC -clib=ansi -lmalloc -lndos -create-app  -o %deliverables%\ZXSpectrum_FIXED.prg %mypath%\display_macros.c %mypath%\powerUps.c %mypath%\enemy.c %mypath%\invincible_enemy.c %mypath%\level.c %mypath%\character.c %mypath%\text.c %mypath%\missile.c %mypath%\strategy.c %mypath%\input.c %mypath%\main.c

Remark: My code is probably OK because it compiles and works fine if compiled with CC65.

I wonder whether this is an interrupt-related bug.

   Fabrizio

zx70, 2017-07-27:
Could you try by compiling at different optimization levels?  Just add -O1 or -O0 to zcc.

Fabrizio-Caruso, 2017-09-14:
Issue solved long ago.

closed 2017-09-14 by Fabrizio-Caruso