z88dk/z88dk #292: conio.h: kbhit()

Fabrizio-Caruso, 2017-07-19:
Current kbhit() implementation is not consistent with Windows/DOS and CC65 implementation:
kbhit() consumes the keyboard input it is supposed to only check for existence.

In such a way, there seems to be no wait to have a portable (conio.h) routine that checks for the existence of keyboard input and do something with it without waiting for it if such input has not been entered.

It would be very nice to have kbhit() behave in a standard way or at least to have a portable solution to interact with the keyboard for all or most targets in Z88DK.

suborb, 2017-07-23:
Good point, I think the best thing for me to do is to remove the macro approach to kbhit()/getch() that we've got and replace with an implementation that has a single character buffer so that if kbhit() has to read the keyboard it's cached and returned on the next getch() - I think this matches the more standard behaviour.

Fabrizio-Caruso, 2017-07-24:
Thanks! Great!
On a different but related topic: it would be great to have conio cprintf/puts run at usable speed in games. 
It would also be great to have conio on the newlib, as well because
conio is de-facto a standard for console input/output and it is the standard portable library for 6502-based compilers.

Fabrizio-Caruso, 2017-07-25:
Will conio ever be ported to the newlib?
I am struggling to port my ANSI C + conio to multipe Z88DK targets due to the lack of usable conio implementation.
And without conio, I will have to implement multiple versions of my program.

suborb, 2017-07-25:
I've implemented custom code for the ZX Spectrum ANSI terminal when running in 32 column mode. It should be a lot quicker now.

If I understand correctly, you're looking a for a single set of control codes that can work on numerous devices?

Fabrizio-Caruso, 2017-07-25:
@suborb, I am writing (just for fun) an 8 bit cross-platform game
(https://github.com/Fabrizio-Caruso/ASCII-CHASE)

The first goal is to have it on as many 8 bit platform as possible (6502 is covered by CC65 and Z80 by Z88DK, other CPUs and a few 6502 and Z80 CPU require special cross compilers).
So, as starting point, I was trying to use conio as default graphics, because conio has made my life so easy with CC65. I thought I could use it with Z88DK but there are many important issues and limitations for the moment. So with Z88DK I am struggling but it is going to be fun once I get the Spectrum working.

A secondary goal is to make specific graphic and sound code for some (or most) of the targets by overriding my the default code.

Since conio.h is problematic, I need alternatives for Z88DK. Ideally portable alternatives but if they do not exist I will have to just code graphics and sound differently for each reasonable target from the very beginning. If possible want to start with something simple (such as character graphics and no sound) and then add fancier effects.

suborb, 2017-07-25:
Ok cool, I think (2) is now sorted for you with the kbhit/getch() changes?

The display side of things is complicated by there being at least 3 ways to get text on screen:

1. classic: Text output (usually firmware)
2. classic: ANSI text driver
3. newlib text output

For some ports (e.g. CPC), the ANSI driver delegates to the firmware output, but for others (e.g. ZX) it's custom code. My gut feel is that it's going to be easier to go down the ANSI driver route and then we can optimise the screen output stage - the ZX one should now be good for you.

Your enhancement approach matches Stefano's when he was enhancing the dstar game.

Fabrizio-Caruso, 2017-07-25:
@suborb, I have just managed to get my game to work on the Spectrum target with
- "print character at x,y" with the classic console directives (as suggested by you)
- "read instantaneous status of the keyboard" with the classic input.h (in_inKey())

This is not the best solution. The best solution for a default skeleton version of my game would be
conio.h because this is the only cross-platform sort-of standard lib and CC65 (as well as Windows) uses it.
conio.h is maybe broken in Z88DK or for some reason my code does not work with it (yet).

I hope conio.h won't be dropped in the newlib and that at least a wrapper version will be created.
Z88DK (and similar projects, e.g., CC65) should aim at portability.

Fabrizio-Caruso, 2017-07-26:
My CONIO wishlist: please make it portable and compatible with other cross compilers and Windows.
I can confirm that the other similar projects (CC65 for the 6502 and CMOC for the  6809)
implement conio extensively and efficiently.

In particular CONIO in these cross compilers (and in  Windows support):
cgetc() and cputc() to handle input/out of single characters.

Please if possible implement a portable and efficient conio interface with the same APIs as in the cross compilers for other architectures.

Fabrizio-Caruso, 2017-07-26:
@suborb, 
I have managed to get my program to work with:
1. in_inKey() from classic input.h to read the keyboard input
2. classic printf with console directives

This is not the optimal solution.
The best solution was conio but it is somehow broken and super slow in Z88DK.
It is a pitty because Z88DK is not portable across its targets.
I wish Z88DK will follow the same path as CMOC for 6809 and CC65 for 6502 and provide efficient conio standard APIs (possibly including the cputc and cgetc commands) for both the classic and the new lib.

suborb, 2017-07-26:
Are you saying that the rewrite (this week) of:

* kbhit/getch 
* The performance boost for -clib=ansi for the ZX Spectrum 32 column mode

don't work? In what way are they failing for you?

Fabrizio-Caruso, 2017-07-26:
@suborb, I have no tried your changes on kbhit, yet.

I do not not what -clib=ansi means. Which APIs? conio? 
If not conio, it is not optimal because it means I have to keep my code non-portable but I will try it anyway to get a performance boost.
If it is not conio, could you point to the API doc? Any example?

I am not familiar with Z88DK's options, libs and compiler combinations.
Could you please provide the exact list of compiler options I should use?

suborb, 2017-07-26:
> @suborb[1], I have no tried your changes on kbhit, yet.

conio in z88dk is really quite thin at the moment, for example cprintf() just wraps printf() etc.
As a result, the behaviour is determined by the underlying terminal driver. Most of the classic platforms have two screen drivers - ANSI terminal and native. The ANSI terminal supports standard vt100 style escape codes and this what you started off using. The native screen drivers typically just call the machines firmware and so use whatever control codes are available there which isn't too useful when it comes to portability.
You started off using -clib=ansi and then switched because the performance of the ZX one wasn't optimal.
So, in summary the change is really just adding -clib=ansi back into the compile line.

Links:

  1. https://github.com/suborb
  2. https://github.com/z88dk/z88dk/issues/292#issuecomment-318032670
  3. https://github.com/

Fabrizio-Caruso, 2017-07-26:
Thanks! 

No, I did start with conio only and nothing else but I got a slow and buggy version of my game.

Then I switched to many things and tried many options without fully understanding what I was doing. Nothing was working correctly. You told me to use classic console directives (or whatever it should be called) and @aralbrec told me what to use in order to get it to compile. I used clib=ansi as one of many attempts.
I am confused by all these different libraries and compiler combinations.

So what does clib=ansi do? Does it switch to other console directives for printf?
Will it break the code that use printf with the console directives?

I do not understand the clib option. It seems it is used to both switch to a different compiler and/or library. I have seen it used with newlib, sdcc_ix, sdcc_yx, ansi,...
Is there a doc page that explains what clib does for each of its possible values? with all its possible values for each of the compilers?

Remark:
I am not happy with the non-portable, non-standard code I am using now but I have to live with it.

zx70, 2017-07-27:
Please be aware that my original idea for conio.h was to take benefit of the many VT ansi option we have on many targets, but it is a very basic converter and it depends on the accuracy of the fputc/fgetc implementation.   Sometimes it is very difficult to get the native functions on the target ROM behave as expected;  this is also the main reason for the VT ANSI emulation existence but the same problem happens on the input side, sometimes the getk() is not implemented (often bouncing to fgetc_cons), sometimes it steals a character from a native buffer and pushing the character back in the right place can be challenging.
My test program for a very minimal conio implementation is snake.c.    I'm adding the gotoxy support to some of the targets where we didn't support the ANSI VT emulation (MSX, SVI, SC3000..) and I noticed already that on some target the result is not ideal (sometimes the cursor is shown, sometimes the keyboard input holds a program waiting for a keypress.   By the way the MSX port looks correct, I'll try to add also the support for colors if it won't be too tricky.

zx70, 2017-07-27:
I forgot to say that the VT 100 compatibility can be checked on the targets table:  
https://github.com/z88dk/z88dk/wiki/Platform

This special text mode needs to be activated by adding the "-pragma-need=ansiterminal" directive to the zcc command.

Fabrizio-Caruso, 2017-07-27:
@zx70 
My code for the Spectrum is using some console directives but I am not sure whether they are the same as the VT100/ANSI ones.

The ones I have used are in
https://www.z88dk.org/wiki/doku.php?id=platform:zx
under
"The standard ZX Spectrum console driver"

Do I need to change my code to use VT100/ANSI directives?
The goal for me is portability at least across different Z88DK targets.

zx70, 2017-07-27:
Fabrizio, you have more chances to have portable code in vt100 mode and on some target platform you will have different text resolution options, but it will cost extra memory.
Moreover at the moment many targets are fully working only in classic mode, which is limited by the sccz80 compiler capabilities.   A quick attempt with conio and vt emulation shouldn't cost too much, I think.

Fabrizio-Caruso, 2017-07-27:
I have finally manage to make my code work with both native and vt100/ANSI directives.

I use different compilers and options, though:

NATIVE DIRECTIVES:
zcc +zx -vn -SO3 -DSPECTRUM_NATIVE_DIRECTIVES -DSPECTRUM_32COL -D__SPECTRUM__ -DAMALLOC -lmalloc -lndos -create-app -o ...

VT100/ANSI:
zcc +zx -SO3 --max-allocs-per-node200000 -startup=1 -clib=sdcc_iy -vn -DSPECTRUM_32COL -D__SPECTRUM__ -DAMALLOC -create-app -o ...

The main code difference is
#if defined(SPECTRUM_NATIVE_DIRECTIVES)
#define gotoxy(x,y) printf("\x16%c%c",y+32,x+32);
#else
#define gotoxy(x,y) printf("\x16%c%c",x+1,y+1);
#endif

Performance-wise they are equivalent. At least I cannot tell the difference while playing the game.
So native directives with the old compiler are doing a decent job but I guess they are less portable. Aren't they?

I would like to port my game to other Z80 computers. I would like to start with MSX and CPC.
Is there a chance that VT100/ANSI could work on these targets?

aralbrec, 2017-07-27:
> VT100/ANSI:
> zcc +zx -SO3 --max-allocs-per-node200000 -startup=1 -clib=sdcc_iy -vn -DSPECTRUM_32COL -D__SPECTRUM__ -DAMALLOC -create-app -o ...
>
> The main code difference is
> #if [...]

This also a native terminal, not a vt100.  After the clutter is remove the compile line should be:

` zcc +zx -SO3 --max-allocs-per-node200000 -startup=1 -clib=sdcc_iy -vn -DSPECTRUM_32COL create-app -o ...`

I assume the "-DSPECTRUM_32COL" is your define otherwise it's not used by the new library.  This is a newlib compile ("-clib=sdcc_iy") using a terminal with 32x24 resolution that understands native control codes.  "-D__SPECTRUM__ -DAMALLOC" are not necessary as "-D__SPECTRUM__" is defined when "+zx" appears on the compile line and "-DAMALLOC" is for the classic library.  The newlib always does automatic initialization of the heap by default.

To use the vt100/ansi driver with the classic library you have to follow [these instructions](https://www.z88dk.org/wiki/doku.php?id=platform:zx#the_vt_ansi_console_driver).

A compile line would look like:

zcc +zx -vn -clib=ansi -pragma-define:ansicolumns=32 -DSPECTRUM_32COL -DAMALLOC -lmalloc -lndos -create-app -o ...

The vt100 sequence for positioning cursor is, I believe:

#define gotoxy(x,y) printf("\x1b[%c;%cf", y+1, x+1)

zx70, 2017-09-14:
I just prepared a gotoxy() implementation for the VT-ansi engine shortcutting to the cursor coordinates  which makes it faster.   It will be in <conio.h> along with a similar trick for the "set VT rendition" related stuff.

zx70, 2017-09-14:
printf-free implementation of "conio" should be online tomorrow morning, let's cross fingers  ;)

zx70, 2017-09-18:
I did quite an extensive test of the new conio support for functions  gotoxy(), vtrendition(), clrscr(), screensize().   They all now seem to behave correctly.
I also tried to reproduce the scroll related issue, but I couldn't find problems.
I've modified ansitest and run it on the Aquarius with the ANSI clib:

  for (x=0; x<11; x++)
  {
    printf ("%c[%u;%uH*\n",27,10+x,25+x);
    printf ("%c[%u;%uH*\n",27,20-x,25+x);
	
	gotoxy(40,24);
	printf (".");
	//printf ("!");
  }

...together with the kbhit() code by Dominic this should close the issue.

closed 2017-09-18 by zx70